Lillifee/raspiCam #13: Plerase add a feature to rotate the camera

HelgeKeck, 2021-12-25:
Please make it possible to rotate the camera

Lillifee, 2021-12-25:
Hey Helge, thanks for your feedback!
I created a new release with the rotation setting. I add the slider to the camera section (other).
[image]

I'm interested in how do you use the app? Do you use it for timelapse photos or videos? 
I saw your timelapse projects. https://github.com/HelgeKeck/uberlapse
Really nice, I have to try it out someday.

HelgeKeck, 2021-12-25:
wow, thats quick. thank you. 

Could you also made the settings persistent please? That would be super nice. So i dont need to set it manually to the last settings.

Im using this to find the perfect camera settings, your interface makes it really easy, i like it. 

i needed the rotation feature bc my camera is mounted sidewards bc my camera cable is only 3 cm long

[image]

[image]

HelgeKeck, 2021-12-25:
i also recommended your project in the RatRig discord server, you dont get enough attention for this awesome project. Put a small screen to it and you have a nice digital camera, well done

Lillifee, 2021-12-25:
That looks amazing ;) congrats, really nice job!
Thanks a lot for your recommendation. I appreciate that!

I put that project a little bit on hold. I was not sure if people were interested in such a simple front-end. In the beginning, I just wanted to observe the 3d printer, take some timelapse pictures from parties, etc.

I also tried a little bit around with TensorFlow. I'm not sure where this project is heading.
[image]

I did another release to persist the settings.
The initial idea was to create different presets for the sections, but that's a good start.

If you mess them up, you can reset them in the top right corner of each section or delete the files in the raspiCam/settings folder.
[image]

Lillifee, 2021-12-26:
What features would you need to capture the pictures directly from raspicam instead of copying and pasting the arguments? Would it help to capture a picture using the gpio pins?

HelgeKeck, 2021-12-26:
As i said, i just use it to find the right settings. For my project this kind of tech/ui, a seperate web ui, isnt helpful at all, it needs to be integrated in Klipper and Mainsail (the 3d printer os) . Settings and controlling needs to be done from gcode and not a web ui. Also, nodejs has a fat ass compared to pure python :-) I try to keep it as simple as possible. But if you are looking for ways to pimprove your project then i would start in the photo gallery, sorting, ordering, fitlering, deleteing, ....... consistent thumbnail sizes, .....

HelgeKeck, 2021-12-26:
i would also switch to ffmepg, this opens up a ton of possibilities for your project, like live youtube streaming, compressed timelapses, .......

Lillifee, 2021-12-26:
Sounds reasonable! Thanks. Maybe i find some time in the next weeks. 😉 Have a nice holiday and a happy new year!

HelgeKeck, 2021-12-26:
You too, and thanks again for making the changes so quick

closed 2021-12-26 by Lillifee