Lillifee/raspiCam #19: Oops! Browser does not support media source extension.

swekley, 2022-06-11:
Hi!

Whenever I try to reach raspiCam from my iPhone I get a plain white page.
In browser console I get the message "Oops! Browser does not support media source extension."

Also tried my sisters iphone, mom's iphone.. Seems like apple doesn't support media source extension.
Is there a way around this that you know of?

I already tried 8 different browsers without any luck..

Lillifee, 2022-06-11:
Hey swekley,

I just went to the jmuxer website and saw the compatibility list. I'm so sorry, I wasn't aware the solution doesn't work with iPhones.
I am quite busy the last few weeks, but I will try to find a solution. 😔

compatible with browsers supporting MSE with 'video/MP4. it is supported on:

Chrome for Android 34+
Chrome for Desktop 34+
Firefox for Android 41+
Firefox for Desktop 42+
IE11+ for Windows 8.1+
Edge for Windows 10+
Opera for Desktop
Safari for Mac 8+

swekley, 2022-06-12:
No worries..
I use my pi to make a timelapse of the growth of strawberries in my climate chamber and eventually I'd like to use raspiCam since[ rpi cam web interface](https://github.com/silvanmelchior/RPi_Cam_Web_Interface) looks like it's last update was in 2004 haha

I also have a suggestion, I really like the FIFO thingy in rpi cam web interface because I can use crontab to shoot a pic every 5 minutes without manually enabling it after every reboot.
When you have the time, could you maybe add something like FIFO, or maybe some kind of api where I could curl to localhost with crontab to take a picture?

I wish I had the time to learn programming languages, I have so many other ideas to extend the features of raspiCam..

Lillifee, 2022-06-13:
Hey sounds nice!!

I'm not so familiar with RPI cam. Do you want to create custom cron jobs to run multiple jobs side by side etc, or simply continue the timelapse after a reboot? I'm asking because raspisill already includes a timelapse mode. Maybe we can just continue the same mode after restart?

I already had a quick look at the iPhone/IOS issue. I'm not an expert in video streaming, but since the iPhone doesn't support media source extension, it looks like I have to switch to webrtc or mjpeg. Sound like I have to rewrite the streaming. I will let you know as soon I find some time to try a little bit around.

swekley, 2022-07-02:
Sorry I've been a bit busy..
The way I create my timelapses now is by shooting a picture every 5 minutes on the pi (through cron) and then compile it to a video on my macbook.
I think the timelapse mode in raspistill does basically the same thing, but I'd prefer having more control and doing it manually through cron.

This is the cron line I use with RPI cam
`*/5 8-23 * * * echo "im" > /var/www/cam/FIFO`
So a simple CLI command that will take a picture within raspiCam would do the job.

For the iphone issue, I guess in my case a more simple (temporarily) solution would be a separate page (/iossucks? lol..) that takes a mid quality picture and shows it in plain html. But of course, for overall wider support webrtc or mjpeg would be best.

Lillifee, 2022-07-23:
Hey swekley,

Raspberry Pi is transitioning from a legacy camera software stack (raspivid, raspistill) to an open-source stack based on libcamera. Before continuing, please check if your raspberry installation is up to date.
You can try to run libcamera-hello to test your version. I switched to the new libcamera apps and adapted to the new command line parameters.

After that, I investigated a little bit into webrtc, but that's quite an overhead and complicated. So I decided to implement MJPEG first. I have no iPhone to try it out, but I hope safari supports my implementation ;) As soon as I have more time and an iPhone, I want to try the broadway player again.

You can switch to MJPEG in the settings menu -> stream -> codec
[image]

Regarding the cron job. I think the easiest option to capture images in raspicam via cron is a simple web request. 
To capture an image with the current settings, you can call /api/start. You can also try it in your browser. It behaves the same as you press the capture button. 

`* * * * * wget -O /dev/null http://localhost:8000/api/start`

The website supports the following commands:
- **/api/start**  start capture
- **/api/stop**  stop capture (e.g. for videos)
- **/api/stream/mjpeg** mjpeg stream
- **/api/stream/live** h264 stream but not viewable from the web
- **/api/gallery** gallery
- **/api/gallery/delete** delete items in gallery

And the different settings via get and post:
- **/api/camera** camera settings
- **/api/preview** previewsettings
- **/api/stream** streamsettings
- **/api/vid** video settings
- **/api/photo** photo settings
- **/api/control** control settings (mode, running etc.)

AntoneWTR, 2023-01-09:
Hi,
I'm having the same issue here.
After switching to MJPEG on my desktop browser, I'm still landing on a blank page on my iphone with safari & chrome.
Is that normal ? Maybe there is a button "save settings" that I missed ?

Lillifee, 2023-01-13:
Hey @GLOUPY-RnD 

Thanks for testing with your iPhone. I didn't get a response from swekley, so I thought it worked for him.
I re-added the Broadway h264 player and adjusted some parts. I will try to get an iPhone this weekend and try it out before making the next release. Thanks a lot. I will reach out to you as soon as the new version is available.

Lillifee, 2023-01-13:
Hey @GLOUPY-RnD 

I created a new release with error handling, the broadway player, and tested it with an iPhone.

You should see this message if you select the **JMuxer** player on the iPhone.
[image]

Then you can select the **MJPEG** codec or use the new **H264 with Broadway player**  and press the **retry button**. 
This player should also work on an iPhone.
[image]

Please let me know if it works for you.

GLOUPY-RnD, 2023-01-13:
Hey @Lillifee, it is working now !
No blank page even with the "wrong" player.
I can also confirm that with the Broadway player it works with both codecs.
_iPhone SE 2020 for me on Chrome (108.0.5359.112) and Safari (latest iOS version to this day)_

I'll open another issue, there is something a little weird but not that important.

Great work !

Lillifee, 2023-01-13:
Nice! Thanks for testing!!!

closed 2023-01-13 by Lillifee