Lillifee/raspiCam #22: Feature Request: Control Motor

ETE-Design, 2022-10-26:
Hi.

Would it be possible to add a way to control a Motor -> Backwards / Forward? Connected to GPIO?

Lillifee, 2022-10-27:
Hey, 

Do you already have a motor wired to the raspberry, or do you set it up all from scratch? May I ask what your use case is? I assume you want to use a stepper motor? There are already stepper driver libraries for the node that we could use. 
https://github.com/nkolban/node-stepper-wiringpi

Right now, I don't have a lot of time for new projects because our child was just born. I don't know when I find time to wire something up and try around. Do you have a little bit of experience with typescript and are willing to help?

ETE-Design, 2022-10-27:
Hi.

I will make it from scratch, I'm an voluntary in a SwimClub... So was thinking to make some DIY Cable Cam's that we can use over each lane when the Club have Competitions 😀 So the motor will move the Cable Cam forward and backwards... Know it is a lot of work, cause have to make 8 of them if I can get it working... Later on would be nice if they could follow the Swimmer by itself... I have almost none experience in programming, but can mabye learn it over time...

Lillifee, 2022-10-28:
Hey, that sounds nice. Is one axis enough, or do you plan for multiple axes to rotate the camera? Maybe you also want to check some camera slider projects to start and test if you get enough speed with a stepper motor.  

https://iotdesignpro.com/projects/raspberry-pi-stepper-motor-control-through-a-webpage-using-flask
https://github.com/topics/camera-slider
https://github.com/akarazeevprojects/StepperProjects

Right now, I don't know when I find time to test a motor with a raspberry pi.

ETE-Design, 2022-10-28:
One Axis should be enough to start with, to not make it to complicated... As said I'm not a programmer, so will have to wait then 😀 A possibility to make follow the swimmer automatic would of cause be the greatest way, but I'm not a programmer... So for now a library for controlling the motor and a button would be fine enough 😍

Lillifee, 2022-12-28:
Hey @ETE-Design,

I created a separate branch to try around with a stepper motor.
https://github.com/Lillifee/raspiCam/tree/stepper

I added a joystick with 2 axes to the web interface.
[image]

With a separate section in the settings to enable and configure the two stepper motors.
[image]

I think it's a good start to hook up the hardware and check if everything works as expected.
As soon as I find more time, I want to work on a smoother acceleration and the bundling of the onoff library. 

Here is the new version:
[raspiCam.zip](https://github.com/Lillifee/raspiCam/files/10315410/raspiCam.zip)

Copy the new version to your raspberry pi and start the server using `node server.js`.
An error should appear `Error: Cannot find module 'onoff'`. To fix this you have to install the onoff library to control the motor using the GPIO pins `npm install onoff`

I tried two different stepper drivers I had lying around. DRV8825 and a TB6600.
You have to use the GPIO pin number. By default, those pins are used, but you can configure them in the web interface.

https://www.rototron.info/raspberry-pi-stepper-motor-tutorial/
https://makersportal.com/blog/raspberry-pi-stepper-motor-control-with-nema-17
[image]

https://danielwilczak101.medium.com/control-a-stepper-motor-using-python-and-a-raspberry-pi-11f67d5a8d6d
[image]

To try it, you could use any stepper motor you have lying around e.g. nema or a cheap 28byj-48. I used a nema17 for my tests.

ETE-Design, 2023-01-02:
@Lillifee Wauv, that looks so nice :-) Have just Orderes parts so I soon can start playing around with it.
I know I'm asking a lot, but is really excited to try this out.

Will it some how later be possible to make a Server / Client to let up to 8-16 X RPI Zero stream their Data / Video to one RPI 4 - 8 GB over WiFi, then I can connect the RPI 4 to Ethernet to access all the cameras remote, and that way not burden each of the Zero to get best possible Quality.
Would like the Zero Software to be as "Light" as possible so it only have to handle Video, the the RPI 4 - 8 GB Can handle all Client Connections.
The RPI 4 Should make it's only WiFi so only the Zero connect to it to keep it simple as possible :-D

Lillifee, 2023-01-02:
Hey,

I'm not sure if this project is the right approach to handle that use case. It's just a web UI written in js with the node using the libcamera apps in the background. This setup is quite an overhead for such an application. The bandwidth is limited and I already had problems streaming and displaying the camera live with a high resolution. 

It sounds like you are looking more a surveillance software with multiple cameras like motion eye or similar. Setup and configure multiple raspberry doesn't really fit into this project and feels more like a complete new project. 
https://pimylifeup.com/raspberry-pi-security-camera/

Since my baby is born, my time is really limited. I dont think I can handle that in my free time.

ETE-Design, 2023-01-07:
Hi.

You'r right it will require some modifications, but really like your GUI, and that it is a lightweight system :-)
The idea is to move the GUI to an Raspberry Pi 4 - 8 GB, that way the Zero W 2 should get more bandwidth... I Came across WebRTC wo focus on low latency with high quality pictures over network.

[https://github.com/paullouisageneau/libdatachannel/](url)

I have also looked at surveillance software, but they seems really bulky and dont know about stepper support...

Congrats with your newborn, they take all your time but it is worth it :-D