Lillifee/raspiCam #25: Gain limited to 4

daingster, 2023-01-03:
Hi, I am using a rpi high quality camera on RPI 3+ (Yes I know, need to upgrade to 4 someday)

The camera is used in a dark room. When I start the camera manually, I set the gain to 20 or it's too dark
libcamera-still --gain 20

The gain in the application is limited between 0 and 4. It's too dark even at 4. Could it be possible to get to a higher value? 

Thank you for your great application, makes a really user-friendly baby monitor.

Lillifee, 2023-01-03:
Hey @daingster

thanks for your comment. The RPI3 is fine as long as the app works. It also has a lower power consumption than the RPI4. 👍

A gain of 20 is really high. Does the camera work without adjustments in daylight? To make sure, there is no hardware problem - did you try to adjust the aperture at your lens? 
https://www.google.com/url?sa=t&source=web&rct=j&url=https://static.raspberrypi.org/files/product-guides/Typical_CS-Mount_Lens_Guide.pdf&ved=2ahUKEwjL36ud5ar8AhXFhP0HHUXYBeUQFnoECBkQAQ&usg=AOvVaw2_35uJP_pArW9eKbmuynBb

If you are using it in complete darkness as a baby monitor and you start with 20. What do you suggest as the upper limit for the slider? I couldn't find the upper limit of the libcamera. Is 30 already completely white in your setup?

Lillifee, 2023-01-03:
Hey @daingster 

I created a new release and increased the gain to 100. 
https://github.com/Lillifee/raspiCam/releases/tag/v1.3.2

It seems the libcamera apps handle the gain differently than the old raspistill.
Thanks for the feedback, and let me know if you encounter more problems.

daingster, 2023-01-09:
Wow that was quick. Thank you so much, works fine now at gain 20.

I am actually using this lens. The aperture is wide open, but still pretty dark. 
https://www.waveshare.com/8-50mm-zoom-lens-for-pi.htm

I find libcamera to render a more fluid preview, but darker than raspistill on my current hardware. The latter had an -ISO parameter that I used all the time.

Thank you for your great software

closed 2023-01-13 by Lillifee