Lillifee/raspiCam #31: Startup Photo Settings missing

GLOUPY-RnD, 2023-01-13:
Hi,
After installing the new release, I started the web page and by default it is in Photo mode but the settings are missing (Timeout, Resolution, Quality, Timelapse).

I had to select video mode and switch back to photo mode in order to make them appear.

Dunno if it is normal so I'm pointing it out ^^

Cheers

Lillifee, 2023-01-13:
Ahhhh sorry for that. By default its not set to Photo. Should be only at the first startup but i will fix it tomorrow. Thanks for reporting that issue!

Lillifee, 2023-01-14:
This bug should be fixed in the latest version.
https://github.com/Lillifee/raspiCam/releases/tag/v1.3.5

GLOUPY-RnD, 2023-01-15:
Hi,
I can confirm that this is fixed :)

closed 2023-01-15 by Lillifee