Lillifee/raspiCam #33: Feature request : Capture with a button

GLOUPY-RnD, 2023-01-15:
Hi,

I started a discussion but I don't know if you've seen it.
Anyway. Your soft is really cool and almost perfectly suits my need !
The only missing part for me would be a way to capture photos with the press of a button.

So may I suggest something like this :
[image]

I first thought that any GPIO for me would work but then I realised that I might not be the only one in need of this feature and depending on their project might need another GPIO.
So I figured that a dropdown list of GPIO would maybe be better ? There is maybe some GPIO that can not be used anyway and it would prevent user mistakes.

I also thought about a debounce time that could be adjusted too. Depending on the switch the user wants to implement it might differ.

Anyway. That would be really great ! :)

Have a good day.
Antoine / GLOUPY

Lillifee, 2023-01-15:
Hey @GLOUPY-RnD 

I saw your discussion, but I had to try the onoff library before I could promise you the implementation. I updated and added the onoff library. Now I should be able to work on the button.

GLOUPY-RnD, 2023-01-15:
Awesome :D

Lillifee, 2023-01-15:
Hey @GLOUPY-RnD 

I published [raspiCam v2.0.0](https://github.com/Lillifee/raspiCam/releases/tag/v2.0.0) including a new GPIO section in the settings.
[image]

The dropdown list to prevent user mistakes was a great idea. I would need to read some documents and try out which GPIO pins are usable. Until now, I added all GPIO pins from 2 to 27. I suggest starting with a GPIO pin without another usage like 17, 27, 22, etc. 
Like all other settings, there is no restart required. It's quite fast to try another pin.

Let me know if that works for you.

I would love to see it in action. Maybe you could post some pictures or even a video of your setup.
https://github.com/Lillifee/raspiCam/discussions/28

GLOUPY-RnD, 2023-01-15:
Wow you're fast !
I'm going to Paris today and will be back home tomorrow. 
I'll check this out as soon as I can !

I'll need a bit more time for the setup, though. But I definitely will !
Thanks a lot.

Lillifee, 2023-01-17:
I knew I run into problems with the onoff library using a precompiled binary. I think I installed node18 on my raspberry pi
```
curl -sL https://deb.nodesource.com/setup_18.x | sudo bash -
sudo apt install nodejs
```

you could also build the epoll.node on your raspberry with your installed version by running 
```npm install epoll``` and copy the epoll.node from /node_modules/build/Release/epoll.node to the raspicam folder.

I'm really sorry for that :(

GLOUPY-RnD, 2023-01-19:
Hi @Lillifee 
took me longer than expected but here I am !

- nodejs 18.x did work, I was able to start the program.

- The combination of Broadway player with H264 codec used to work on iphone but not anymore (MJPEG still works fine)

- I'm not able to take a photo with a button. It is maybe due to my wiring ?
I selected GPIO 17 in your soft and plugged a button between GPIO 17 (pin 11) and Ground (pin 14).

I thought about two things:
1- For newbies like me, maybe a correspondence table should be added :

[image]

It would prevent wrong wiring. And maybe some prefer referencing the pins by their physical position ? Anyway just a suggestion.

2- Some might need pull up, other pull down. So maybe adding another setting like "connected to : ground / 5v / 3.3v" might be a good idea ? But it might be redundant with your "rising / falling" option ?

Finally, I did get an error after resetting the "GPIO module" and selecting GPIO 17 again :

[image]

Hope that help :)
Cheers.

Lillifee, 2023-01-19:
Hey @GLOUPY-RnD 

Thanks a lot for your feedback.

- Nice that node 18 work for you. i really need to find a solution for the precompiled binary
- Do you get an error with the broadway player?
- I'm not that experienced with the gpio on the raspberry. All sites i found pull the pin to the 3.3v https://www.google.com/amp/s/raspberrypihq.com/amp/use-a-push-button-with-raspberry-pi-gpio/
- The rising/falling is used to trigger on press/release the button
- The last error is strange. It seems node had no permission to write the settings.

I'm right now at the hospital and back home at the weekend. I hope you get the button to work with the current version...

GLOUPY-RnD, 2023-01-19:
Hello @Lillifee

- The Broadway player with H264 codec used to work on iphone but not anymore (MJPEG still works fine) : I don't get any error message, I just have a black square with no live feed.

- I tried to connect it to 3.3v with a resistor and it's working ! Just had to know it. Maybe you should right it somewhere or :

- All GPIO pins have internal pull-up or pull-down resistors that can be controlled from your code.

So basically a switch can be plugged directly from a GPIO to Ground or 5V without the need of an external resistor. That is why I suggested the possibility for the user to choose **from** (GPIO) and **where to** (ground / 5V) in your soft. It changes which pull-up/down internal resistor to choose in your code. But you can also take a stand and tell the user to only connect it to ground if you want ^^

More info here : https://github.com/raspberrypilearning/physical-computing-guide/blob/master/pull_up_down.md

Don't worry, there is no rush !
Hope you're alright.

Lillifee, 2023-01-20:
Hey @GLOUPY-RnD 

I will try to get an iPhone to test again with the broadway player.

I found a section in the onoff library I use. https://github.com/fivdi/onoff - Configuring Pullup and Pulldown Resistors. It's possible to reconfigure it, but I have no chance to try it here 
Since I need this epoll precompiled binary, I'm not 100% happy with the library. Maybe I can find another library that offers to configure the behavior in the library.

GLOUPY-RnD, 2023-01-20:
Hey,

this is to far advanced for me there ^^
It is working for me now. As long a you write it somewhere that the selected GPIO must be connected to the 3.3v with a 220ohms resistor it should be fine. 
It is just more convenient to not have resistors for the average user but it's not that difficult to buy either.

I'm going to the next step which is finding a way to turn on and off your soft with the press of a button ! ^^

Cheers

Lillifee, 2023-01-20:
hey, 

Yeah lets see if i find another package to access the gpios. But to be honest, i left the resistor away for my tests 😂

Whats the exact reason to turn on/off the raspicam using a button? To start up and shut down the raspberry pi, or do you have another conflicting software running? You could auto-start raspicam using a service like described in the rreadme. Then you could power on/off the raspberry?

GLOUPY-RnD, 2023-01-20:
Yo,

So if you remember I'm using the Raspberry for my 3D printer a Raise3D Pro3.
The pro 3 model comes with a stock camera that is basically useless...

I'd like to achieve two things (which are almost finished btw !) :
- A better live feed (hopefully iOS compatible)
- A timelapse mode

At first I thought using the HD pi camera for both but quickly realised that the camera can only be used by one program at a time. And not being able to monitor live when I do a timelapse could be harmful.
So I found a usb webcam for the live feed as it doesn't need to be very high quality and installed [this soft](https://github.com/brutella/hkcam).
**First point : Done !**

Now for the time lapses, I don't intend to use it all the time. In order to achieve a "good" looking time lapse, I need to config my printer for an extra step (moving the head to X.Y position for it to trigger a switch) between every layers. It will increase the printing time. And once I have done it for a piece, there is no point doing it for all the copies that I'll make. Customers often orders 10-15pieces of the same object.
Anyway this is why I need to config a button or something to turn on and off your soft. There is no point for it running all the time for nothing.
**Second point : in the making !**

BTW I did config a service that check pins 5-6 for auto on/off of the PI. I've connected a relay to the USB of the printer. When I turn on the printer it starts the pi. Same when I turn it off.

GLOUPY-RnD, 2023-05-17:
I finally have time to work again on this project !
I'm having difficulties triggering a shot with a button.
I'll investigate more this matter tomorrow and let you know what I find.

Lillifee, 2023-05-20:
Hello @GLOUPY-RnD,

I have kept this issue open to enhance the pin selection (physical/gpio). However, I suggest closing this particular issue and creating a new one for any additional matters.

closed 2023-05-20 by Lillifee