Lillifee/raspiCam #37: Error running raspiCam

Morcegolas, 2023-02-07:
When I run raspiCam I get this errors, can you help me please?

`
~/raspiCam $ node server.js
/home/pi/raspiCam/server.js:2
(()=>{var e={8967:e=>{function t(e){var t=new Error("Cannot find module '"+e+"'");throw t.code="MODULE_NOT_FOUND",t}t.keys=()=>[],t.resolve=t,t.id=8967,e.exports=t},4410:(e,t,n)=>{"use strict";var a=n(6690),i=n(436);function o(e){if(!(this instanceof o))return new o(e);this.headers=e.headers,this.negotiator=new a(e)}function r(e){return-1===e.indexOf("/")?i.lookup(e):e}function s(e){return"string"==typeof e}e.exports=o,o.prototype.type=o.prototype.types=function(e){var t=e;if(t&&!Array.isArray(t)){t=new Array(arguments.length);for(var n=0;n<t.length;n++)t[n]=arguments[n]}if(!t||0===t.length)return this.negotiator.mediaTypes();if(!this.headers.accept)return t[0];var a=t.map(r),i=this.negotiator.mediaTypes(a.filter(s)),o=i[0];return!!o&&t[a.indexOf(o)]},o.prototype.encoding=o.prototype.encodings=function(e){var t=e;if(t&&!Array.isArray(t)){t=new Array(arguments.length);for(var n=0;n<t.length;n++)t[n]=arguments[n]}return t&&0!==t.length?this.negotiator.encodings(t)[0]||!1:this.neg

SyntaxError: Unexpected token .
    at Module._compile (internal/modules/cjs/loader.js:760:23)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:827:10)
    at Module.load (internal/modules/cjs/loader.js:685:32)
    at Function.Module._load (internal/modules/cjs/loader.js:620:12)
    at Function.Module.runMain (internal/modules/cjs/loader.js:877:12)
    at internal/main/run_main_module.js:21:11
`

Lillifee, 2023-02-07:
Hey @Morcegolas

Thanks a lot for your feedback. I created a new release, including a switch from ESM to commonJs, and removed the precompiled onoff binary.
I hope this fixes your problem with the cjs-loader. 

https://github.com/Lillifee/raspiCam/releases/tag/v2.0.2

Can you reinstall the raspiCam and try it again?
Please let me know if the new version works for you.

Morcegolas, 2023-02-07:
Hello, I tried with the new version but got this error:

```
~/raspiCam $ node server.js
/home/pi/raspiCam/server.js:13
}`)(t,Fi,this,e,i);return r}function Q0(t,e,n){if(!t||typeof t!="object"&&typeof t!="function")throw new TypeError("argument obj must be object");var a=Object.getOwnPropertyDescriptor(t,e);if(!a)throw new TypeError("must call property on owner object");if(!a.configurable)throw new TypeError("property must be configurable");var i=this,r=Ai(),o=An(r[1]);o.name=e,"value"in a&&(a=L0(t,e,n));var s=a.get,p=a.set;typeof s=="function"&&(a.get=function(){return Fi.call(i,n,o),s.apply(this,arguments)}),typeof p=="function"&&(a.set=function(){return Fi.call(i,n,o),p.apply(this,arguments)}),Object.defineProperty(t,e,a)}function eu(t,e,n){var a=new Error,i;return Object.defineProperty(a,"constructor",{value:eu}),Object.defineProperty(a,"message",{configurable:!0,enumerable:!1,value:e,writable:!0}),Object.defineProperty(a,"name",{enumerable:!1,configurable:!0,value:"DeprecationError",writable:!0}),Object.defineProperty(a,"namespace",{configurable:!0,enumerable:!1,value:t,writable:!0}),Objec

SyntaxError: Unexpected token ?
    at Module._compile (internal/modules/cjs/loader.js:760:23)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:827:10)
    at Module.load (internal/modules/cjs/loader.js:685:32)
    at Function.Module._load (internal/modules/cjs/loader.js:620:12)
    at Function.Module.runMain (internal/modules/cjs/loader.js:877:12)
    at internal/main/run_main_module.js:21:11
```

Lillifee, 2023-02-07:
Hey @Morcegolas 

Which node version do you have running? 
```
node --version
```

Can you try to install node 18 or newer?
```
curl -sL https://deb.nodesource.com/setup_18.x | sudo bash -
sudo apt install nodejs
```

Maybe some features are no longer supported with the old version.
If you have node 18 already installed, can you try to clone and build it on your raspberry pi?

```
git clone https://github.com/Lillifee/raspiCam.git

cd raspiCam/

npm install
npm run build
```

Morcegolas, 2023-02-07:
Thanks for helping. 
I’m using version 11.15.0 because newer versions are not supported by armv6l, only if I install from not official sources. I will try and let here the result

Lillifee, 2023-02-07:
Oh ok, that makes sense. 
I did not re-test it with the old node version. I will try to fix it!

Morcegolas, 2023-02-07:
With version 19 it starts but gives some erros:

[server] starting services...
[control] starting Stream ...
[process] libcamera-vid --nopreview --width 1280 --height 720 --framerate 25 --bitrate 10000000 --codec H264 --timeout 0 --profile baseline --inline --output -
[button] onoff library not found...
To enable GPIO support please install the onoff library and restart raspiCam. 'npm install onoff'
 Error [ERR_MODULE_NOT_FOUND]: Cannot find package 'onoff' imported from /home/pi/raspiCam/server.js
[server] server listening on 8000
[0:07:57.514712115] [1370]  INFO Camera camera_manager.cpp:299 libcamera v0.0.3+40-9b860a66
[0:07:57.605410027] [1375] ERROR V4L2 v4l2_device.cpp:352 'imx477 10-001a': Unable to set controls: Device or resource busy
[0:07:57.822794023] [1375]  INFO RPI raspberrypi.cpp:1425 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media3 and ISP device /dev/media0
[0:07:57.826200944] [1370]  INFO Camera camera.cpp:841 Pipeline handler in use by another process
ERROR: *** failed to acquire camera /base/soc/i2c0mux/i2c@1/imx477@1a ***

Lillifee, 2023-02-07:
But this sounds like the camera is already in use. Can you stop it and try to run libcamera without raspicam. Maybe another process use the camera.

Morcegolas, 2023-02-07:
How do I stop it? 
If I run ‘libcamera-still -o test.jpg’ and I open the test.jpg I can see the picture that camera take.

Lillifee, 2023-02-07:
Hey,

I could not find a lot about this error *** failed to acquire camera /base/soc/i2c0mux/i2c@1/imx477@1a *** 
But this error is coming from libcamera. Can you try to run the same command as raspicam. 

```
libcamera-vid --nopreview --width 1280 --height 720 --framerate 25 --bitrate 10000000 --codec H264 --timeout 0 --profile baseline --inline -o test.h264
```

Regarding node 10 - I bundled raspicam and tried it on my raspberry zero with node 10.24.0
[raspiCam.tar.gz](https://github.com/Lillifee/raspiCam/files/10675421/raspiCam.tar.gz)

Morcegolas, 2023-02-07:
I didn’t test the last command you sent me, if I run vcgencmd get_camera this is what it retuns:
supported=0 detected=0, libcamera interfaces=0

Morcegolas, 2023-02-07:
libcamera-vid --nopreview --width 1280 --height 720 --framerate 25 --bitrate 10000000 --codec H264 --timeout 0 --profile baseline --inline -o test.h264
[1:34:34.583982383] [2372]  INFO Camera camera_manager.cpp:299 libcamera v0.0.3+40-9b860a66
[1:34:34.668619853] [2373] ERROR V4L2 v4l2_device.cpp:352 'imx477 10-001a': Unable to set controls: Device or resource busy
[1:34:34.879615531] [2373]  INFO RPI raspberrypi.cpp:1425 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media3 and ISP device /dev/media0
[1:34:34.883144509] [2372]  INFO Camera camera.cpp:841 Pipeline handler in use by another process
ERROR: *** failed to acquire camera /base/soc/i2c0mux/i2c@1/imx477@1a ***

Lillifee, 2023-02-07:
> I didn’t test the last command you sent me, if I run vcgencmd get_camera this is what it retuns: supported=0 detected=0, libcamera interfaces=0

i get the same response using 
vcgencmd get_camera
supported=0 detected=0, libcamera interfaces=0

libcamera-vid --nopreview --width 1280 --height 720 --framerate 25 --bitrate 10000000 --codec H264 --timeout 0 --profile baseline --inline -o test.h264
[0:21:00.021051732] [3249]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3700-f30ad033
[0:21:00.461334481] [3253]  WARN RPI raspberrypi.cpp:1252 Mismatch between Unicam and CamHelper for embedded data usage!
[0:21:00.466640430] [3253]  INFO RPI raspberrypi.cpp:1368 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media3 and ISP device /dev/media0
[0:21:00.478312317] [3249]  INFO Camera camera.cpp:1029 configuring streams: (0) 1280x720-YUV420
[0:21:00.494070165] [3253]  INFO RPI raspberrypi.cpp:759 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 1920x1080-SBGGR10_1X10 - Selected unicam format: 1920x1080-p

I will try to update my zero to the latest version, but that can take a while
``` 
sudo apt update
sudo apt upgrade
```

Lillifee, 2023-02-07:
with the updated version 

[0:39:51.220626091] [9305]  INFO Camera camera_manager.cpp:299 **libcamera v0.0.3+40-9b860a66**
[0:39:51.386452034] [9310]  WARN RPI raspberrypi.cpp:1308 Mismatch between Unicam and CamHelper for embedded data usage!
[0:39:51.394977288] [9310]  INFO RPI raspberrypi.cpp:1425 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media3 and ISP device /dev/media0
Mode selection:
    SRGGB10_CSI2P 640x480 - Score: 6093.33
    SRGGB10_CSI2P 1640x1232 - Score: 4557.83
    SRGGB10_CSI2P 1920x1080 - Score: 3250
    SRGGB10_CSI2P 3280x2464 - Score: 12887.7
    SRGGB8 640x480 - Score: 7093.33
    SRGGB8 1640x1232 - Score: 5557.83
    SRGGB8 1920x1080 - Score: 4250
    SRGGB8 3280x2464 - Score: 13887.7
[0:39:51.484152947] [9305]  INFO Camera camera.cpp:1028 configuring streams: (0) 1280x720-YUV420 (1) 1920x1080-SBGGR10_CSI2P
[0:39:51.489418104] [9310]  INFO RPI raspberrypi.cpp:805 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 1920x1080-SBGGR10_1X10 - Selected unicam format: 1920x1080-pBAA

Lillifee, 2023-02-07:
Hey @Morcegolas 

Does libcamera-vid without arguments work?
```
libcamera-vid -o test.h264
```
Maybe you could drill down the error by removing one argument after the other and figure out which one caused libcamera-vid to fail.

You could also try to get more logs using LIBCAMERA_LOG_LEVELS=0 in front of the command.
```
LIBCAMERA_LOG_LEVELS=0 libcamera-vid ....
```

Please let me know if you figure out what the problem is.

Morcegolas, 2023-02-07:
pi@rpivpn:~ $ LIBCAMERA_LOG_LEVELS=0 libcamera-vid -o test.h264
Preview window unavailable
[3:46:50.192308999] [3703] DEBUG IPAModule ipa_module.cpp:329 ipa_rpi.so: IPA module /usr/lib/arm-linux-gnueabihf/libcamera/ipa_rpi.so is signed
[3:46:50.195561978] [3703] DEBUG IPAManager ipa_manager.cpp:245 Loaded IPA module '/usr/lib/arm-linux-gnueabihf/libcamera/ipa_rpi.so'
[3:46:50.197148968] [3703] DEBUG IPAModule ipa_module.cpp:329 ipa_vimc.so: IPA module /usr/lib/arm-linux-gnueabihf/libcamera/ipa_vimc.so is signed
[3:46:50.198596959] [3703] DEBUG IPAManager ipa_manager.cpp:245 Loaded IPA module '/usr/lib/arm-linux-gnueabihf/libcamera/ipa_vimc.so'
[3:46:50.199993950] [3703]  INFO Camera camera_manager.cpp:299 libcamera v0.0.3+40-9b860a66
[3:46:50.202022937] [3704] DEBUG Camera camera_manager.cpp:108 Starting camera manager
[3:46:50.239650698] [3704] DEBUG DeviceEnumerator device_enumerator.cpp:224 New media device "bcm2835-codec" created from /dev/media2
[3:46:50.241499686] [3704] DEBUG DeviceEnumerator device_enumerator_udev.cpp:95 Defer media device /dev/media2 due to 5 missing dependencies
[3:46:50.253486610] [3704] DEBUG DeviceEnumerator device_enumerator_udev.cpp:320 All dependencies for media device /dev/media2 found
[3:46:50.254892601] [3704] DEBUG DeviceEnumerator device_enumerator.cpp:252 Added device /dev/media2: bcm2835-codec
[3:46:50.257452585] [3704] DEBUG DeviceEnumerator device_enumerator.cpp:224 New media device "bcm2835-isp" created from /dev/media0
[3:46:50.258763577] [3704] DEBUG DeviceEnumerator device_enumerator_udev.cpp:95 Defer media device /dev/media0 due to 4 missing dependencies
[3:46:50.261718558] [3704] DEBUG DeviceEnumerator device_enumerator.cpp:224 New media device "bcm2835-isp" created from /dev/media1
[3:46:50.262966550] [3704] DEBUG DeviceEnumerator device_enumerator_udev.cpp:95 Defer media device /dev/media1 due to 4 missing dependencies
[3:46:50.273824481] [3704] DEBUG DeviceEnumerator device_enumerator_udev.cpp:320 All dependencies for media device /dev/media0 found
[3:46:50.276039467] [3704] DEBUG DeviceEnumerator device_enumerator.cpp:252 Added device /dev/media0: bcm2835-isp
[3:46:50.288641387] [3704] DEBUG DeviceEnumerator device_enumerator_udev.cpp:320 All dependencies for media device /dev/media1 found
[3:46:50.290986372] [3704] DEBUG DeviceEnumerator device_enumerator.cpp:252 Added device /dev/media1: bcm2835-isp
[3:46:50.296908334] [3704] DEBUG DeviceEnumerator device_enumerator.cpp:224 New media device "unicam" created from /dev/media3
[3:46:50.300637311] [3704] DEBUG DeviceEnumerator device_enumerator_udev.cpp:95 Defer media device /dev/media3 due to 2 missing dependencies
[3:46:50.306616273] [3704] DEBUG DeviceEnumerator device_enumerator_udev.cpp:320 All dependencies for media device /dev/media3 found
[3:46:50.308045264] [3704] DEBUG DeviceEnumerator device_enumerator.cpp:252 Added device /dev/media3: unicam
[3:46:50.312254237] [3704] DEBUG Camera camera_manager.cpp:151 Found registered pipeline handler 'PipelineHandlerRPi'
[3:46:50.315777215] [3704] DEBUG DeviceEnumerator device_enumerator.cpp:312 Successful match for media device "unicam"
[3:46:50.318800195] [3704] DEBUG DeviceEnumerator device_enumerator.cpp:312 Successful match for media device "bcm2835-isp"
[3:46:50.324724158] [3704] DEBUG V4L2 v4l2_device.cpp:635 'imx477 10-001a': Control: Exposure (0x00980911)
[3:46:50.327740139] [3704] DEBUG V4L2 v4l2_device.cpp:635 'imx477 10-001a': Control: Horizontal Flip (0x00980914)
[3:46:50.329827125] [3704] DEBUG V4L2 v4l2_device.cpp:635 'imx477 10-001a': Control: Vertical Flip (0x00980915)
[3:46:50.331932112] [3704] DEBUG V4L2 v4l2_device.cpp:635 'imx477 10-001a': Control: Camera Orientation (0x009a0922)
[3:46:50.333316103] [3704] DEBUG V4L2 v4l2_device.cpp:635 'imx477 10-001a': Control: Camera Sensor Rotation (0x009a0923)
[3:46:50.333760100] [3704] DEBUG V4L2 v4l2_device.cpp:635 'imx477 10-001a': Control: Vertical Blanking (0x009e0901)
[3:46:50.335145092] [3704] DEBUG V4L2 v4l2_device.cpp:635 'imx477 10-001a': Control: Horizontal Blanking (0x009e0902)
[3:46:50.335555089] [3704] DEBUG V4L2 v4l2_device.cpp:635 'imx477 10-001a': Control: Analogue Gain (0x009e0903)
[3:46:50.336914080] [3704] DEBUG V4L2 v4l2_device.cpp:635 'imx477 10-001a': Control: Red Pixel Value (0x009e0904)
[3:46:50.337323078] [3704] DEBUG V4L2 v4l2_device.cpp:635 'imx477 10-001a': Control: Green (Red) Pixel Value (0x009e0905)
[3:46:50.338370071] [3704] DEBUG V4L2 v4l2_device.cpp:635 'imx477 10-001a': Control: Blue Pixel Value (0x009e0906)
[3:46:50.340481058] [3704] DEBUG V4L2 v4l2_device.cpp:635 'imx477 10-001a': Control: Green (Blue) Pixel Value (0x009e0907)
[3:46:50.341925049] [3704] DEBUG V4L2 v4l2_device.cpp:635 'imx477 10-001a': Control: Pixel Rate (0x009f0902)
[3:46:50.342335046] [3704] DEBUG V4L2 v4l2_device.cpp:635 'imx477 10-001a': Control: Test Pattern (0x009f0903)
[3:46:50.343821036] [3704] DEBUG V4L2 v4l2_device.cpp:635 'imx477 10-001a': Control: Digital Gain (0x009f0905)
[3:46:50.344405033] [3704] ERROR V4L2 v4l2_device.cpp:352 'imx477 10-001a': Unable to set controls: Device or resource busy
[3:46:50.353491975] [3704] DEBUG CameraSensor camera_sensor.cpp:374 'imx477 10-001a': No static test pattern map for 'imx477'
[3:46:50.380202805] [3704] DEBUG IPAManager ipa_manager.cpp:303 IPA module /usr/lib/arm-linux-gnueabihf/libcamera/ipa_rpi.so signature is valid
[... cut ...]
[3:46:50.603323388] [3704] DEBUG Camera camera_manager.cpp:151 Found registered pipeline handler 'SimplePipelineHandler'
[3:46:50.604164383] [3704] DEBUG Camera camera_manager.cpp:151 Found registered pipeline handler 'PipelineHandlerUVC'
[3:46:50.604529381] [3704] DEBUG Camera camera_manager.cpp:151 Found registered pipeline handler 'PipelineHandlerVimc'
[3:46:50.605694373] [3703]  INFO Camera camera.cpp:841 Pipeline handler in use by another process
ERROR: *** failed to acquire camera /base/soc/i2c0mux/i2c@1/imx477@1a ***

Morcegolas, 2023-02-07:
Even without arguments I’m getting the same error.

Morcegolas, 2023-02-07:
Ok, I'm going to do a clear install, and only run lib camera-vid to see.
After that if it fails again I'll try their GitHub.
Thanks for trying to help me out!
Thumbs up!

Morcegolas, 2023-02-07:
Although the live video is not working I tried to take a photo and it worked, but after that it stopped, only the first photo worked and after that when  tried to take another photo it returned:
`[control] starting Video ...
[process] libcamera-vid --nopreview --width 1920 --height 1080 --framerate 25 --bitrate 10000000 --timeout 0 --output /home/pi/raspiCam/photos/2023-02-07T19-16-14-324Z.h264
[control] starting Stream ...
[process] libcamera-vid --nopreview --width 1280 --height 720 --framerate 25 --bitrate 10000000 --codec H264 --timeout 0 --profile baseline --inline --output -
[0:02:47.858003903] [730]  INFO Camera camera_manager.cpp:299 libcamera v0.0.3+40-9b860a66
[0:02:48.027558177] [731]  INFO RPI raspberrypi.cpp:1425 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media3 and ISP device /dev/media0
[0:02:48.035524492] [731] ERROR V4L2 v4l2_videodevice.cpp:897 /dev/video16[17:cap]: Unable to set format: Invalid argument
[0:02:48.037329603] [731] ERROR RPI raspberrypi.cpp:933 Failed to set format on ISP stats stream: 0x0-BSTA
[0:02:48.047732768] [731] ERROR V4L2 v4l2_videodevice.cpp:897 /dev/video16[17:cap]: Unable to set format: Invalid argument
[0:02:48.049510848] [731] ERROR RPI raspberrypi.cpp:933 Failed to set format on ISP stats stream: 0x0-BSTA
[0:02:48.059184159] [731] ERROR V4L2 v4l2_videodevice.cpp:897 /dev/video16[17:cap]: Unable to set format: Invalid argument
[0:02:48.062612168] [731] ERROR RPI raspberrypi.cpp:933 Failed to set format on ISP stats stream: 0x0-BSTA
[0:02:48.070490381] [731] ERROR V4L2 v4l2_videodevice.cpp:897 /dev/video16[17:cap]: Unable to set format: Invalid argument
[0:02:48.072276469] [731] ERROR RPI raspberrypi.cpp:933 Failed to set format on ISP stats stream: 0x0-BSTA
Mode selection:
    SRGGB10_CSI2P 1332x990 - Score: 4377.47
    SRGGB12_CSI2P 2028x1080 - Score: 2314.5
    SRGGB12_CSI2P 2028x1520 - Score: 3717.7
    SRGGB12_CSI2P 4056x3040 - Score: 4604.7
[0:02:48.099440234] [730]  INFO Camera camera.cpp:1028 configuring streams: (0) 1280x720-YUV420 (1) 2028x1080-SBGGR12_CSI2P
[0:02:48.103937493] [731]  INFO RPI raspberrypi.cpp:805 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1080-SBGGR12_1X12 - Selected unicam format: 2028x1080-pBCC
[0:02:48.108223505] [731] ERROR V4L2 v4l2_videodevice.cpp:897 /dev/video16[17:cap]: Unable to set format: Invalid argument
[0:02:48.112216174] [731] ERROR RPI raspberrypi.cpp:933 Failed to set format on ISP stats stream: 0x0-BSTA
ERROR: *** failed to configure streams ***
`

Morcegolas, 2023-02-07:
https://ibb.co/RgFR3B8
https://ibb.co/4Vtfqq5

Here you can see the photo, its buried because its dark and I didn't focus the camera yet, and the second one eis just the error. 
I upgraded node vo v18.9.1 from unofficial builds but its the same :/

Lillifee, 2023-02-08:
Hmm yeah, those are kernel panic errors coming from raspberry and libcamera. I really dont know whats wrong with your setup.
I think you will get the same errors with libcamera-vid.

Maybe something is wrong with your hardware? I hope its just the SD card like in the following problem: 
https://raspberrypi.stackexchange.com/questions/101536/apt-get-not-working-on-raspberry-pi-zero-w-internal-error-oops-17-1-arm

you could also try to clock the raspi to 700mhz to get more stability?
https://forums.raspberrypi.com/viewtopic.php?t=282481

Morcegolas, 2023-02-08:
I’ll try to install on my old RPI 3 to see if I can get it to work. Thanks

Morcegolas, 2023-02-08:
Ok, it’s working with my RPI 3B. When you mentioned hardware it clicked, my Zero had some issues when I bought it and never worked well if I ask from it. Thanks for all your help

closed 2023-02-08 by Morcegolas

Lillifee, 2023-02-09:
Nice that it works with the rpi3. 
I'm sorry for your zero 😔 Maybe you can get a zero2 soon 👍

What do you think? Is it worth to keep the software backwards compatible or does the unofficial node versions work good enough?