Lillifee/raspiCam #38: Running on raspberry pi3?

wociscz, 2023-02-07:
Hi, in the doc there is info installing it on rpi4. Tried to install it on rpi3b. Everything went ok, I am able to capture still or video with `libcamera-jpeg` and `libcamera-vid`.
When I run raspicam, I'm getting only black rectangle even tho camera is properly initialized. Is it known to not working on previous rpis?

output:
```
[server] starting services...
[control] starting Stream ...
[process] libcamera-vid --nopreview --width 1280 --height 720 --framerate 25 --bitrate 10000000 --codec H264 --timeout 0 --profile baseline --inline --output -
[0:34:02.577385629] [1812]  INFO Camera camera_manager.cpp:299 libcamera v0.0.3+40-9b860a66
[server] server listening on 8000
[0:34:02.738569305] [1813]  INFO RPI raspberrypi.cpp:1425 Registered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device /dev/media3 and ISP device /dev/media0
Mode selection:
    SRGGB10_CSI2P 1536x864 - Score: 3100
    SRGGB10_CSI2P 2304x1296 - Score: 3400
    SRGGB10_CSI2P 4608x2592 - Score: 25592.8
[0:34:02.747652903] [1812]  INFO Camera camera.cpp:1028 configuring streams: (0) 1280x720-YUV420 (1) 1536x864-SBGGR10_CSI2P
[0:34:02.748063008] [1813]  INFO RPI raspberrypi.cpp:805 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 1536x864-SBGGR10_1X10 - Selected unicam format: 1536x864-pBAA
```

[image]

wociscz, 2023-02-07:
Ok, my bad. When I try to switch to mjpeg stream it works without any problem.

closed 2023-02-07 by wociscz

Lillifee, 2023-02-07:
Hey @wociscz 

h264 should work as well. What OS do you use? Mac have a problem with jmuxer because of the media source extension. You could try to switch the h264 player from jmuxer to broadway in the expander. 

Can you also check the console output for errors in your browser? In most browsers you can open the developer tools using F12.

wociscz, 2023-02-07:
as per doc - raspberry os lite, fully upgraded.

Lillifee, 2023-02-07:
Sorry i ment your os and the browser you access your raspberry pi. The h264 player runs in the browser.

wociscz, 2023-02-07:
Linux mint (ubuntu22 based) Firefox 109-01

wociscz, 2023-02-07:
Weird. It is working now with h264 stream also. But choppy/blocky for 1080p. Ok for 720p. Seems rpi3b is not powerful enough. Don't mind this issue. Thanks.

wociscz, 2023-02-09:
Just to confirm, Firefox is kinda weird with raspicam. Sometimes it works, sometime don't (but without any errors in dev tools). Chromium and Chrome is works like a charm all the time.

Lillifee, 2023-02-09:
Thanks for testing. I'll try it out!