Lillifee/raspiCam #42: Frequency out-of-sync grey bands on captured images

gns-jdaniel, 2023-03-15:
Is there any way to add an option to add an anti-flicker adjustment for capturing still images? The photos I'm getting under these 60 Hz grow lights have rolling horizontal grey bands on them. 

See attached samples...
[image]
[image]
[image]
[image]

Lillifee, 2023-03-16:
Hey @gns-jdaniel ,

Unfortunately, it seems that there is no "--flicker" setting available in the libcamera apps, and I couldn't find a replacement for it when using libcamera-still to capture images. This has also been mentioned in the "Differences compared to raspicam" section of the  [raspberrypi documentation](https://www.raspberrypi.com/documentation/computers/camera_software.html#differences-compared-to-raspicam-apps).

I came across an open issue on the [libcamera-apps](https://github.com/raspberrypi/libcamera-apps/issues/444) Github repository that addresses this problem and it might be worth checking the  [raspberrypi forum](https://forums.raspberrypi.com/viewforum.php?f=43&sid=b8c50044b5ef6db1d88e705fd5cca810) to see if anyone has found a workaround for this issue.

If we can find a setting for libcamera-still, I can try adding it to the Raspicam app. Let me know if you have any suggestions or insights on this matter.

Lillifee, 2023-03-24:
In the previous thread, you mentioned that flicker can be reduced by not using the timelapse function. I was wondering if you could test this with the new version and share your results?

Unfortunately, I can't do anything until libcamera re-adds the flicker setting.

gns-jdaniel, 2023-03-29:
This setting?

`extern const std::array<const ControlValue, 2> [LensShadingMapModeValues](https://libcamera.org/api-html/namespacelibcamera_1_1controls_1_1draft.html#aa16e02d987ec168b4d910a6e30a5aa79);
 extern const [Control<int32_t>](https://libcamera.org/api-html/classlibcamera_1_1Control.html) [LensShadingMapMode](https://libcamera.org/api-html/namespacelibcamera_1_1controls_1_1draft.html#a0849d24af45ddc950e44637abc7c0e01);
 enum [SceneFlickerEnum](https://libcamera.org/api-html/namespacelibcamera_1_1controls_1_1draft.html#a97800fb1d3afef07eb550873af0dbe16) {
     [SceneFickerOff](https://libcamera.org/api-html/namespacelibcamera_1_1controls_1_1draft.html#a97800fb1d3afef07eb550873af0dbe16ae1ba6904ca34f985055ad682a7fafb10) = 0,
     [SceneFicker50Hz](https://libcamera.org/api-html/namespacelibcamera_1_1controls_1_1draft.html#a97800fb1d3afef07eb550873af0dbe16a47ff793173200c8e77e95cc5aa3cfb8d) = 1,
     [SceneFicker60Hz](https://libcamera.org/api-html/namespacelibcamera_1_1controls_1_1draft.html#a97800fb1d3afef07eb550873af0dbe16aee10a6ec1f8e8ba14d3977c34a996850) = 2,
 };`

Lillifee, 2023-03-29:
I'm using the libcamea-still command line tool. Maybe this settings are  only internal used? Can you try to call libcamera-still with the flicker setting on the command line? 

In the Raspberry pi documtation they also mention they don't support this setting in libcamera apps
https://www.raspberrypi.com/documentation/computers/camera_software.html#differences-compared-to-raspicam-apps
"There is no support for setting the flicker period yet."

I will try to update my system and check if there is a setting for it.

gns-jdaniel, 2023-03-29:
I've been playing with the --shutter setting, in hopes of finding an ideal value in microseconds that matches my 60Hz lighting.

This is what my camera is capable of:
`$ libcamera-still --list-cameras
Available cameras
-----------------
0 : imx708 [4608x2592] (/base/soc/i2c0mux/i2c@1/imx708@1a)
    Modes: 'SRGGB10_CSI2P' : 1536x864 [120.13 fps - (768, 432)/3072x1728 crop]
                             2304x1296 [56.03 fps - (0, 0)/4608x2592 crop]
                             4608x2592 [14.35 fps - (0, 0)/4608x2592 crop]`

If I use a command like this:
`libcamera-still --hdr --rotation 180 --sharpness 0 --brightness 0 --ev 0 --gain 0 --metering average --exposure normal --awb auto --lens-position 4.6 --nopreview --quality 80 --width 4056 --height 3040 --timeout 2000 --thumb 320:240:35 --shutter 25000 --output test-25000ms-01.jpg`

where I find some magical value for --shutter...do you think this will work?

AND in case anyone reading this can assist...any idea what a good value would be? 14.35fps/60Hz = 239166 microseconds, so I've obviously done my math wrong...but am I on the right track? Do you think I can find a reasonable value for shutter that will deliver a flicker-free image?