Lillifee/raspiCam #43: Disable auto-preview files?

gns-jdaniel, 2023-03-15:
I get a low-res xyz-0000-preview1.jpg file for every xyz-0000.jpg file that's captured - could you please add an option to disable this extra file?

Lillifee, 2023-03-16:
Hey @gns-jdaniel 

Exiv2 is utilized to generate the thumbnail. If you want to disable the preview image, you can simply remove Exiv2.

I can add a flag to disable it. However, it's important to note that the preview file is used to display a thumbnail on the website.  Would you like to completely disable the preview image or is it sufficient to move it to another folder instead?

Lillifee, 2023-03-24:
I added a new setting in the control section to turn on/off the thumbnail extraction.

[image]

gns-jdaniel, 2023-03-29:
Perfect! Thank you!