Lillifee/raspiCam #46: Add date/time to file metadata

gns-jdaniel, 2023-03-15:
Apparently, "adding the -x flag gets EXIF data written to the image, the EXIF data includes the date and time"...could you please add this as an option when capturing, so I can plug the timestamp into the frame using FFMPEG?

I've constructed the following command, which grabs the filename (which I originally thought was the timestamp before realizing that every file was named the same) and burns it in yellow on each frame, but I'd ideally like the actual timestamp rather than the index number (the [0-9][0-9][0-9][0-9] glob was to omit the -preview1 files that currently get auto-created):

ffmpeg -y -r 24 -export_path_metadata 1 -pattern_type glob -i './*[0-9][0-9][0-9][0-9].jpg' -vf "drawtext=fontfile=/usr/share/fonts/truetype/freefont/FreeSerif.ttf:x=20:y=20:fontsize=40:fontcolor=yellow:text='%{metadata\:lavf.image2dec.source_basename\:NA}'" -c:v libx264 video-file.mp4

Lillifee, 2023-03-16:
Hey @gns-jdaniel 

While libcamera allows users to set EXIF information, I don't think this feature will work with the libcamera timelapse feature. This is because every image in the timelapse series would have the same date/time which is not ideal. :( 
https://www.raspberrypi.com/documentation/computers/camera_software.html#still-command-line-options

Is it possible with ffmpeg to use the date/time from the file instead?
However, i may need to check the amount of effort required to create a custom timelapse feature and trigger the capture from raspicam in the meantime.

Lillifee, 2023-03-24:
Hey @gns-jdaniel,

I noticed that most of the EXIF data already exists without any additional input. However, I wanted to add an extra EXIF field, specifically 'EXIF.DateTimeOriginal', to my image using the following command:

```
libcamera-still --quality 80 --width 4056 --height 3040 --timeout 2000 --exif EXIF.DateTimeOriginal=someData --output /home/pi/raspiCam/photos/123.jpg
```

Despite adding this field, I was unable to extract the EXIF data along with the metadata using ffmpeg. 

However, I have created a new release that includes the EXIF data (including 'EXIF.DateTimeOriginal') and also allows for turning off thumbnail extraction. Additionally, I have added the ability to create a new timelapse using the cron syntax. As a result, the images contain the accurate date and time information, reflecting the time when they were captured.

https://github.com/Lillifee/raspiCam/releases/tag/v2.0.5

Please let me know if you find out how to use the exif data in ffmpeg.
Thanks

gns-jdaniel, 2023-03-29:
My new ffmpeg generate-timelapse-video command pulled the file creation timestamp and added it in yellow in the top left corner of the screen - I'll have to modify it now that the filename format has changed.

`ffmpeg -y -r 24 -export_path_metadata 1 -pattern_type glob -i './*[0-9][0-9][0-9][0-9].jpg' -vf "drawtext=fontfile=/usr/share/fonts/truetype/freefont/FreeSerif.ttf:x=20:y=20:fontsize=40:fontcolor=yellow:text='%{metadata\:DateTime\:def_value}'" -c:v libx264 video-file.mp4`

Thank you for adding the EXIF data to the frames, and switching to cron. I was still having an issue where raspicam seemed to hang after a few hundred images, and I had to log in and restart the timelapse capture after 350-400 frames (ish).

Lillifee, 2023-03-29:
I hope it works with the new cron timelapse without restart. Please let me know.

I'm not sure what i did wrong last week but the DateTimeOrignal works.
```
text='%{metadata\:DateTimeOriginal}'
```

closed 2023-04-02 by Lillifee