Lillifee/raspiCam #47: Detect blank frames and pause capture? Is there a 50-image limit to timelapse?

gns-jdaniel, 2023-03-15:
My timelapse capture currently seems to stop at exactly 50 images - how do I get this to capture indefinitely?

And it seems that the lights in the room where I'm capturing are on a schedule and the room goes pitch-black for a number of hours. Rather than capturing hundreds of black frames, is it possible to pause capturing when an entirely black frame is detected, then resume when the image changes from black?

I know this is a tall order, but just wondering... :)

dpoulson, 2023-03-20:
Just to add more info, if using PNG as the output, I get 0-52, using JPG I get 0-63.

gns-jdaniel, 2023-03-20:
@dpoulson : Interesting that you're seeing a wider range with JPG. I'm only outputting JPG files, and I consistently get 49-51.

dpoulson, 2023-03-20:
Yeah, constantly 0-51 or 52 for png, and 0-63 with jpg. Changing resolution has no difference, nor does changing the number of seconds between each photo (tried 3 and 10 second intervals). Always stops at 51 or 52 for png, or 63 for jpg. 

Just tried it on another raspiCam module I've put together and having the same issue. 

Module 1 is a PiZero2 (Camera module 3 Wide), second one is a PiZeroW (Camera module v1.3). Both running 32bit bullseye with nodejs 18.9.1.

Lillifee, 2023-03-20:
Hey,

Thank you for providing the settings and conducting the investigations. 
I believe I have identified the issue and have created a new release.

https://github.com/Lillifee/raspiCam/releases/tag/v2.0.4

Please let me know if it works for you.

I'm planning to replace the built-in timelapse feature with an implementation in raspicam and start with a simple chron setting. This change would provide a few advantages, such as enabling the video preview between captures, eliminating our dependence on the filename of libcamera, and allowing for more advanced timer settings (e.g., capturing only during daytime)

dpoulson, 2023-03-20:
Yep, just built from source with that new patch and it looks like its working. At least its got past the 63rd image and is still going. 

Thanks for the rapid response! :D I need these camera modules for Star Wars Celebration in a few weeks. (testing them this weekend at another convention hopefully)

Switching to a chron based system sounds good. Unfortunately I'm not much of a javascript programmer so can't help there but happy to be a tester.

Lillifee, 2023-03-20:
Star Wars celebration? You make me curious :D
Maybe you want to share your usage and some pics in the discussions?  
https://github.com/Lillifee/raspiCam/discussions/28

It's awesome to see how you guys use raspicam. It also helps to keep the motivation up ;)

gns-jdaniel, 2023-03-20:
That's wonderful news, @Lillifee ! Thank you for your quick response!

There's a whole set of Grade 5 students who will be thrilled that their hydroponics lab timelapses will no longer be missing chunks of growth spurts!

I also noticed that the straight libcamera-still snapshot DOESN'T have the rolling grey bands from a sync issue with the gro-lights, so maybe switching to cron vs libcamera's built-in timelapse feature will fix that annoying problem with the current frames.

dpoulson, 2023-03-20:
Hopefully if it works at MegaCon (birmingham UK) this weekend, I'll post my test up on the usage discussion thread. Just a pi zero with camera and a big ol' powerbank to track visitors to the Droidbuilders booth.

gns-jdaniel, 2023-03-20:
Hmmm...getting a 404 error when trying to download...

--2023-03-20 12:46:07--  https://github.com/Lillifee/raspiCam/releases/latest/download/raspiCam.tar.gz
Resolving github.com (github.com)... 140.82.113.3
Connecting to github.com (github.com)|140.82.113.3|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://github.com/Lillifee/raspiCam/releases/download/v2.0.4/raspiCam.tar.gz [following]
--2023-03-20 12:46:08--  https://github.com/Lillifee/raspiCam/releases/download/v2.0.4/raspiCam.tar.gz
Reusing existing connection to github.com:443.
HTTP request sent, awaiting response... 404 Not Found
2023-03-20 12:46:08 ERROR 404: Not Found.

gns-jdaniel, 2023-03-20:
Yup works fine now - thanks again for the amazingly quick response!

Lillifee, 2023-03-24:
Does the new release resolve your issues?

The new timelapse should also work for more than 50 images. I hope the new cron syntax works for you.

Regarding the black frames:
You could try the new release and define a cron job that runs from e.g. 8-18 for example [0 8-18 * * *](https://crontab.guru/#0_8-18_*_*_*)
or by deleting the black images by size ```find -size -100k -delete```

gns-jdaniel, 2023-03-29:
Unfortunately, I've upgraded raspiCam during the "lights-out" time in the hydroponics lab...I'll have to see how it's doing when the lights turn back on again. Unfortunately, the all-black frames turn out to be larger than the lit frames because the camera is creating a very highly-defined image with zillions of tiny pixel "snow" like the attached (651KB instead of the usual 490-510KB, depending on the rolling horizontal flicker/sync bars)
[image]

gns-jdaniel, 2023-03-29:
photo.json says this:

`{"quality":80,"width":4056,"height":3040,"timeout":2000,"timelapse":300000,"timelapseTimeout":0}`

Lillifee, 2023-03-29:
hmm... 

can you try to remove the photo.json. The timelapse and the timelapseTimeout should not be used anymore. Those settings should also vanish, when you change one of the photo settings like the resolution.

If it still not work, could you send me the output of the raspicam in the console when you capture an image? If you run it as a service you can check the output by 
```journalctl -u raspicam```

gns-jdaniel, 2023-03-29:
Deleting photo.json seemed to fix it - now it's working. Thank you!

closed 2023-04-02 by Lillifee