Lillifee/raspiCam #52: Feature request : "Select all" photos

GLOUPY-RnD, 2023-05-17:
Hi Lilifee,

A button for selecting all photos would be nice (or a direct "delete all" button).

[image]

Side note : you can see that thumbnails does not work with png file format. Or maybe the resolution is too big ? Not that it matters to me but I'm reporting the issue if it matters for others.

[image]

Lillifee, 2023-05-20:
Hey @GLOUPY-RnD,

I admit I haven't spent much time in the gallery, but I'll definitely look into it and see what I can do.

Regarding the PNG format, it's unfortunate that the thumb feature in libcamera-still is only available for JPG.
So far, I've only used JPG as the image format. Could you please let me know the reason behind choosing PNG over JPG? If PNG is preferred, I would need to create the thumbnail manually and possibly integrate an additional library for that purpose.

Lillifee, 2023-05-21:
Hey @GLOUPY-RnD 

The latest release now allows to select a full day or all images on the top right.

GLOUPY-RnD, 2023-05-23:
Working fine here, too !
Still no thumbnails, though ;)

closed 2023-05-23 by Lillifee