Lillifee/raspiCam #53: Upgrade : Soft version

GLOUPY-RnD, 2023-05-17:
Hi Lilifee,

I updated to the latest version of your soft. Well I think I did.
A version written somewhere, maybe at the bottom of the setting tab, would be a good idea in order to check that it's been updated.

Just a thought.

Lillifee, 2023-05-20:
Hey @GLOUPY-RnD,

thanks for your feedback. I'll try to add it in the next version.

Lillifee, 2023-05-21:
Hey @GLOUPY-RnD 

I added the verison in the application settings section and on startup.

[image]

[image]

closed 2023-05-23 by Lillifee