Lillifee/raspiCam #54: Help needed: Snapshot address

Morcegolas, 2023-05-20:
- I use Raspicam to use with my Octoprint instance, for the Stream Address everything works as expected, I use http://hqrpi.local:8000/api/stream/mjpeg and its working, but to use with octolapse I also need the snapshot address, and the default is {camera_address}?action=snapshot that translates to http://hqrpi.local:8000/?action=snapshot but it doesn't work. Can you help me with this, or it's just not possible?

Thanks.

Lillifee, 2023-05-20:
Hey @Morcegolas,

Thank you for reaching out. To capture an image, you can give this URL a try: 
http://hqrpi.local:8000/api/start.

I'm not very familiar with OctoPrint, so I'm not sure what Octolapse expects for the snapshot. 
Does it require any specific format and returning the image? I'll see if i can find out more and give it a shot.

Morcegolas, 2023-05-20:
> Hey @Morcegolas,
>
> Thank you for reaching out. To capture an image, you can give this URL a try: http://hqrpi.local:8000/api/start
>
> I'm not very familiar with OctoPrint, so I'm not sure what [...]

If I use that it just starts taking snapshots and instead of getting the snapshot image I get an message "ok", I need the link for the current snapshot to use in octolapse.

Thanks.

Lillifee, 2023-05-20:
Ok, i'll see what i can do. 
Thanks

Lillifee, 2023-05-21:
Hi @Morcegolas,

I wanted to let you know that I've added two new API commands in the latest release of the application. https://github.com/Lillifee/raspiCam/releases/tag/v2.0.7

Here are the details of the new commands:

- **/api/capture** - This command captures an image and returns it to the user.
- **/api/stream/mjpeg/snapshot** - This command takes a snapshot of the MJPEG stream. It is a faster option as it only captures the last image of the stream. However, please note that the MJPEG stream needs to be running for this command to work properly.

I would greatly appreciate it if you could try out both commands and provide me with feedback on whether they are functioning as expected.

Thank you very much!

Morcegolas, 2023-05-21:
Thanks for helping out. I'm getting this error in octolapse when using **/api/stream/mjpeg/snapshot** because the other one don't work there.
`Camera Test Failed
Errors were detected - An invalid schema was detected while connecting to hqrpi.local:8000/api/stream/mjpeg/snapshot for the 'HQ Camera Network' camera profile.`

This is the help page:
`Webcam Snapshot Address Template
You can either enter the full path for your webcam's snapshot page, or you can use the {camera_address} template. Using the {camera_address} token is preferred if for no other reason than to verify that the base address is correct. The base address is used for custom image preferences as well as some error checking, and it's important that you get it correct.

The default value is {camera_address}?action=snapshot, which works well for mjpg_streamer (bundled with octopi). If you are using Yawcam you probably want to use {camera_address}out.jpg

The default full url after replacing the {camera_address} token with the default 'Base Address' above would be would be http://127.0.0.1:8080/?action=snapshot for mjpg_streamer and http://127.0.0.1:8888/out.jpg for Yawcam.

If the 'Test' button does not work, you can also enter a full snapshot url here. If the URL works in a browser window and returns a JPEG image, it will likely work in Octolapse.`

Morcegolas, 2023-05-21:
This is the log:

`Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_octolapse/camera.py", line 578, in _test_web_camera
    r = retry_session.get(url, stream=True, timeout=timeout_seconds)
  File "/home/pi/oprint/lib/python3.7/site-packages/requests/sessions.py", line 600, in get
    return self.request("GET", url, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/requests/sessions.py", line 587, in request
    resp = self.send(prep, **send_kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/requests/sessions.py", line 695, in send
    adapter = self.get_adapter(url=request.url)
  File "/home/pi/oprint/lib/python3.7/site-packages/requests/sessions.py", line 792, in get_adapter
    raise InvalidSchema(f"No connection adapters were found for {url!r}")
requests.exceptions.InvalidSchema: No connection adapters were found for '10.1.1.27:8000/api/capture'`

Lillifee, 2023-05-21:
Hey, 

thanks for the fast response. Did the api request work in your browser?

This sounds it should work if it's returning a jpeg
"If the URL works in a browser window and returns a JPEG image, it will likely work in Octolapse."

But sounds i have to install octoprint and try a little bit around.

Morcegolas, 2023-05-21:
The api request worked in the browser, the both of them, but none in octolapse :/

Lillifee, 2023-05-21:
Hey @Morcegolas,

I've installed OctoPrint with Octolapse, but I don't have a printer connected yet to conduct a complete test. 
However, I managed to test the camera using the following settings.

You need to provide either the full URL or use the arguments as follows:

**Base address:** http://192.168.3.58:8000/
**Snapshot address (full):** http://192.168.3.58:8000/api/stream/mjpeg/snapshot
**or Snapshot address (with variable name)**: {camera_address}api/stream/mjpeg/snapshot

[image]

During my tests, the MJPEG snapshot always worked, but the api/capture only worked when I reduced the timeout and image resolution. It seems that Octolapse doesn't wait as long for the image. Maybe it's only during test webcam and we could increase the timeout during a real print.

Settings worked for me with the api/capture:

**Base address:** http://192.168.3.58:8000/
**Snapshot address:** {camera_address}api/capture

Raspicam photo settings
**Timeout:** 100ms
**Resolution:** <=5MP

[image]

[image]

Morcegolas, 2023-05-21:
Thanks, it's strange it started working, maybe needed a reboot I don't know. 
Already making a Timelapse of 15h with it to test, I'm using MJPEG because of stability, lower quality but it's working better. 

Since I'm using raspiCam I never got this to work, until yesterday that I open here a question, many thanks. 

I'm also having a problem of having to reboot a lot my raspiCam, since the beginning, and it's powering it off/on because it becomes very unresponsive. I suspect that as I use it with multiple devices/apps (octoprint web, iOS app) and I have two routers with the same Wi-Fi names, as is always jumping between them maybe it overloads the raspberry I don't know.

Anyway, this problem is solved.

closed 2023-05-21 by Morcegolas

Lillifee, 2023-05-22:
Hey @Morcegolas 

i had a similar problem yesterday where i had to reboot. Maybe i broke something in the mjpeg streaming in one of the latest releases. I will review them and test a little bit more. If you find a way to easy reproduce, let me know. 

Thanks

Lillifee, 2023-05-23:
Hey @Morcegolas,

I believe I've identified a possible reason for the unresponsiveness of the raspicam. In one of the recent updates, I made enhancements to the h264 streaming. However, as a consequence, the mjpeg streams are not being closed correctly. This could be the cause of multiple devices running simultaneously, resulting in numerous lingering mjpeg streams in the background.

I have addressed this issue in version 2.0.8, which can be found at https://github.com/Lillifee/raspiCam/releases/tag/v2.0.8.

Morcegolas, 2023-05-23:
Thanks a lot, yesterday I made two big prints, the first one had 5 failed of 35 photos, the second print it didn't take any photo. 

I will update to this version and give you some feedback tomorrow.

Morcegolas, 2023-05-23:
I couldn't install because when I run wget it gave me an error, but I build it from source and it's running the latest build now, I'll do some tests and let you know.

Lillifee, 2023-05-23:
Uh i forgot to add the compiled verison to the release.
Thanks for the info, i fixed it.

Morcegolas, 2023-05-23:
> Uh i forgot to add the compiled verison to the release.
> 
> Thanks for the info, i fixed it.

I thought the problem was on my side, so instead of complaining I compiled it and updated. 
So far so good, no faults and no delays. I'm not using Octolapse because I was in the middle of a 15h print when I updated the raspiCam and I cannot turn it on in the middle of a print but tomorrow I'll try that also. But from the stability and the speed that I'm seeing in the last 6h after multiple local connections and some remote I'm confident that it will work fine. I can't wait to test the capture from the camera and not the mjpeg capture so I can have the full size ;)

Thanks one more time for being so helpful.

Lillifee, 2023-05-24:
I'm uncertain if the full capture has been improved. The capture process in libcamera-still takes a considerable amount of time, and if it exceeds a certain duration, octolapse fails. I'm unsure if this behavior persists during printing. Perhaps we could initiate a test print with a timeout of 100ms and low resolution. Then we can gradually increase the values during the print to identify the limits and determine if the behavior remains consistent.

The latest release primarily focuses on enhancing the stability and performance of the mjpeg streamer. Unfortunately, the capture process still requires the same amount of time.

If you have the time, it would be great if you could share your setup and how you utilize raspicam. While uploading videos of such lengthy timelapse prints might not be feasible, some photos would be greatly appreciated. https://github.com/Lillifee/raspiCam/discussions/28

Lillifee, 2023-05-24:
I stumbled upon a screenshot on the Octolapse website. Perhaps you could attempt adjusting the snapshot timeout in the general settings and give it another try using the api/camera. I didn't come across this particular setting during my own tests.
[image]