Lillifee/raspiCam #57: Stream during recording

Lillifee, 2023-06-12:
Keep the stream active during recording.
Feature request from disucussion https://github.com/Lillifee/raspiCam/discussions/40

Currently, libcamera-vid only supports a single stream at a time. This stream is recorded to a file during recordings and streamed live with different settings on the website. However, if we attempt to preview a 1080p video with a high bitrate, it will cause delays and lags in the live stream. Consequently, it becomes quite challenging to simultaneously record a high-quality video and stream it. In my opinion, the most viable solution would be to save the high-quality video to a file, compress it using ffmpeg, and then stream the compressed version on the website. However, implementing this solution may require significant refactoring since raspicam does not currently utilize ffmpeg.

Lillifee, 2023-06-12:
hmm... would be nice to have a little bit more control over libcamera... 
Hope i find a little bit time in the next week to test ffmpeg...

That's fascinating! :) 
Can the PIR sensor detect those distant bats? They seem to be quite far away, don't they?

james-dmu, 2023-06-12:
I haven't actually tried the PIR with them so I am not sure but I suspect it won't pick them up as their bodies are only around 3-5cm long. For these videos I am just manually recording for a bit and then picking out the best bits. :)