Lillifee/raspiCam #63: My CM4 comes with a touch screen, can I operate these settings on Raspberry Pi?

yq3w, 2023-09-16:
My CM4 also comes with a touch screen. Can I see a preview and operate these settings on Raspberry Pi? Because I can easily focus and adjust aperture, it looks like a great camera.

Lillifee, 2023-09-16:
Hey @yq3w,

Absolutely! You can just open the browser on your Raspberry Pi.
Ensure you have the Raspberry Pi Desktop version installed, then open your web browser and navigate to http://localhost:8000/.

For a more focused setup, consider running it in kiosk mode (browser-only) with automatic website launch on startup. Check out the instructions here: [Raspberry Pi Kiosk Mode Tutorial](https://www.raspberrypi.com/tutorials/how-to-use-a-raspberry-pi-in-kiosk-mode/).

Alternatively, if you prefer a minimalistic approach, install the Raspberry Pi Lite version, which comes with a lightweight desktop environment. You can find the setup details here: [Setting up Raspberry Pi Lite for Kiosk Mode](https://die-antwort.eu/techblog/2017-12-setup-raspberry-pi-for-kiosk-mode/).

The website is optimized for smartphones, it should function smoothly on a small touchscreen too.
Let me know if you encounter any issues or have any further questions.

yq3w, 2023-09-20:
@Lillifee Thank you very much. Everything went smoothly during the installation on the lte system!

Lillifee, 2023-09-20:
Great! I'm delighted that everything is functioning smoothly. Would you mind sharing details about the CM4 with touchscreen you're using? If possible, could you share some pictures of your setup?

yq3w, 2023-09-20:
[image]
[image]
[image]
I don't have a shell yet, but I will design one in the future.

Lillifee, 2023-09-20:
It's already looking fantastic! Please keep me updated on your project's progress and the shell. Looks really nice! Maybe you could share the usage on the discussions page  https://github.com/Lillifee/raspiCam/discussions/28

closed 2023-09-20 by Lillifee