Lillifee/raspiCam #66: Run RaspiCam as a service - files unavailable if not using GitHub clone

billjuv, 2023-12-30:
I installed raspiCam using the following command:

"mkdir raspiCam && cd raspiCam
wget https://github.com/Lillifee/raspiCam/releases/latest/download/raspiCam.tar.gz
tar -xvzf raspiCam.tar.gz"

instead of using a GitHub clone and found the systemd files to be copied were unavailable

"pi@RaspiCam1:~ $ sudo cp raspicam.service /etc/systemd/system
cp: cannot stat 'raspicam.service': No such file or directory"

I finally worked out how to get these files installed and working properly, but certainly not seamless like the rest of the install.

ALSO, I feel "Check the [raspicam.service](https://github.com/Lillifee/raspiCam/blob/master/systemd/raspicam.service) in the systemd folder in the repository and adapt it to your needs." needs further explanation in the documentation. I'm not terribly experienced so "adapt it to your needs" had me puzzled. I finally assumed it should work with the default installation as written so copied as it was. I'm also assuming it's working as I was able to get a snapshot after rebooting without restarting node server.js manually.

I'll share more about needs/configuration when I get a chance

Lillifee, 2023-12-30:
Hey @billjuv 

I agree. I'll fix the service description to match the rest of the documentation soon. Thanks for letting me know!

Lillifee, 2024-01-02:
Hey @billjuv 

I updated the readme:

To automatically start the RaspiCam on startup, you can run the RaspiCam as a service.
Download the [raspicam.service](https://github.com/Lillifee/raspiCam/blob/master/systemd/raspicam.service) from the the repository:

```
wget https://raw.githubusercontent.com/Lillifee/raspiCam/master/systemd/raspicam.service
```

Change the working directory and user details in a text editor like nano or vi if raspicam is stored elsewhere or if a different user is being used.

```
WorkingDirectory=/home/pi/raspiCam
User=pi
```
Copy the raspicam.service file to the systemd:

```
sudo cp raspicam.service /etc/systemd/system
```

... the rest should be ok.

What do you think?

billjuv, 2024-01-03:
Much better! I’ve always been a “drag & drop” person until I started playing with Pis, et al, a couple of times ago (“playing” being the operative word). I am slowly getting back to learning the basics as needed.

Regards,
Bill

Lillifee, 2024-01-03:
Yeah, same here. 
Maybe i can find the time to install the service from the website...
However, thanks for the feedback! 👍

closed 2024-01-03 by Lillifee