Lillifee/raspiCam #67: No stream/image is displayed

hb9thf, 2024-01-01:
Hi,
I've installed RaspiCam on my  Pi 5, with the camera v3 module but when I  start the server no image is displayed and an error appear :

```
[server] version 2.0.10
[server] starting services...
[control] starting Stream ...
[process] libcamera-vid --camera 0 --hdr --exposure sport --lens-position 10.2 --nopreview --width 1280 --height 720 --framerate 25 --bitrate 6000000 --codec H264 --timeout 0 --profile baseline --inline --output -
[server] server listening on 8000
[1:33:43.925141017] [4070]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+118-563cd78e
[1:33:43.939259011] [4078]  INFO RPI pisp.cpp:653 libpisp version v1.0.2 fa44a258644a 22-11-2023 (21:59:22)
[1:33:43.956707129] [4078]  INFO RPI pisp.cpp:1112 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a to CFE device /dev/media0 and ISP device /dev/media2 using PiSP variant BCM2712_C0
[1:33:43.958024911] [4070]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+118-563cd78e
[1:33:43.971881627] [4081]  INFO RPI pisp.cpp:653 libpisp version v1.0.2 fa44a258644a 22-11-2023 (21:59:22)
[1:33:43.989824172] [4081]  INFO RPI pisp.cpp:1112 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a to CFE device /dev/media0 and ISP device /dev/media2 using PiSP variant BCM2712_C0
[1:33:43.990596989] [4070]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format Y16
[1:33:43.990625804] [4070]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format RGB6
[1:33:43.990633397] [4070]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format BGR6
[1:33:43.990642249] [4070]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format PC1M
Mode selection for 1280:720:12:P(25)
    SRGGB10_CSI2P,2304x1296/30.0021 - Score: 1400
Stream configuration adjusted
[1:33:43.990953083] [4070]  INFO Camera camera.cpp:1183 configuring streams: (0) 1280x720-YUV420 (1) 2304x1296-BGGR16_PISP_COMP1
[1:33:43.991039731] [4081]  INFO RPI pisp.cpp:1396 Sensor: /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a - Selected sensor format: 2304x1296-SBGGR10_1X10 - Selected CFE format: 2304x1296-PC1B
ERROR: *** libav: no such profile baseline ***
```

If you take a look on 4th line ([PROCESS]) you will see something seems to be missing after "--output -" 

Kind Regards

Lillifee, 2024-01-01:
Hey @hb9thf 

the output stream is used within the application, so the --output - should be fine. 

The error seams to be related to the --profile baseline. Could you try to copy the command and run it on the command line using --output test.mp4 to save the stream to a file and run it once with the --profile baseline and once without? We have to figure out wich arguments are failing on the rpi5.

`libcamera-vid --camera 0 --hdr --exposure sport --lens-position 10.2 --nopreview --width 1280 --height 720 --framerate 25 --bitrate 6000000 --codec H264 --timeout 0 --profile baseline --inline --output test.h264`

`libcamera-vid --camera 0 --hdr --exposure sport --lens-position 10.2 --nopreview --width 1280 --height 720 --framerate 25 --bitrate 6000000 --codec H264 --timeout 0 --inline --output test.h264`

Additionally, I came across a forum thread discussing the Pi 5:
https://forums.raspberrypi.com/viewtopic.php?t=360283
Unfortunatly i have no rpi5 to test it. 🙁

You could also try to switch to the mjpeg stream to check if the camera works there.

hb9thf, 2024-01-01:
Hey @Lillifee 

I've tested the first command answer exactly the same error : 

```
$ libcamera-vid --camera 0 --hdr --exposure sport --lens-position 10.2 --nopreview --width 1280 --height 720 --framerate 25 --bitrate 6000000 --codec H264 --timeout 0 --profile baseline --inline --output test.h264
[4:01:39.131751852] [4672]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+118-563cd78e
[4:01:39.145953168] [4675]  INFO RPI pisp.cpp:653 libpisp version v1.0.2 fa44a258644a 22-11-2023 (21:59:22)
[4:01:39.163705847] [4675]  INFO RPI pisp.cpp:1112 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a to CFE device /dev/media0 and ISP device /dev/media2 using PiSP variant BCM2712_C0
[4:01:39.164927295] [4672]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+118-563cd78e
[4:01:39.178893962] [4678]  INFO RPI pisp.cpp:653 libpisp version v1.0.2 fa44a258644a 22-11-2023 (21:59:22)
[4:01:39.197011475] [4678]  INFO RPI pisp.cpp:1112 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a to CFE device /dev/media0 and ISP device /dev/media2 using PiSP variant BCM2712_C0
[4:01:39.198555720] [4672]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format Y16
[4:01:39.198589498] [4672]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format RGB6
[4:01:39.198602831] [4672]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format BGR6
[4:01:39.198616924] [4672]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format PC1M
Mode selection for 1280:720:12:P(25)
    SRGGB10_CSI2P,2304x1296/30.0021 - Score: 1400
Stream configuration adjusted
[4:01:39.198844073] [4672]  INFO Camera camera.cpp:1183 configuring streams: (0) 1280x720-YUV420 (1) 2304x1296-BGGR16_PISP_COMP1
[4:01:39.198920962] [4678]  INFO RPI pisp.cpp:1396 Sensor: /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a - Selected sensor format: 2304x1296-SBGGR10_1X10 - Selected CFE format: 2304x1296-PC1B
**ERROR: *** libav: no such profile baseline *****
```
Then, always as normal user the second test command, without profile, return a right problem : 
```
 $ libcamera-vid --camera 0 --hdr --exposure sport --lens-position 10.2 --nopreview --width 1280 --height 720 --framerate 25 --bitrate 6000000 --codec H264 --timeout 0 --inline --output test.h264
[4:04:04.398244085] [4697]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+118-563cd78e
[4:04:04.412399885] [4700]  INFO RPI pisp.cpp:653 libpisp version v1.0.2 fa44a258644a 22-11-2023 (21:59:22)
[4:04:04.430363936] [4700]  INFO RPI pisp.cpp:1112 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a to CFE device /dev/media0 and ISP device /dev/media2 using PiSP variant BCM2712_C0
[4:04:04.431488458] [4697]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+118-563cd78e
[4:04:04.445323275] [4703]  INFO RPI pisp.cpp:653 libpisp version v1.0.2 fa44a258644a 22-11-2023 (21:59:22)
[4:04:04.463279179] [4703]  INFO RPI pisp.cpp:1112 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a to CFE device /dev/media0 and ISP device /dev/media2 using PiSP variant BCM2712_C0
[4:04:04.463999033] [4697]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format Y16
[4:04:04.464035273] [4697]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format RGB6
[4:04:04.464043107] [4697]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format BGR6
[4:04:04.464052033] [4697]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format PC1M
Mode selection for 1280:720:12:P(25)
    SRGGB10_CSI2P,2304x1296/30.0021 - Score: 1400
Stream configuration adjusted
[4:04:04.464263552] [4697]  INFO Camera camera.cpp:1183 configuring streams: (0) 1280x720-YUV420 (1) 2304x1296-BGGR16_PISP_COMP1
[4:04:04.464351700] [4703]  INFO RPI pisp.cpp:1396 Sensor: /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a - Selected sensor format: 2304x1296-SBGGR10_1X10 - Selected CFE format: 2304x1296-PC1B
[libx264 @ 0x5556007c5de0] using cpu capabilities: ARMv8 NEON
[libx264 @ 0x5556007c5de0] profile High, level 3.1, 4:2:0, 8-bit
Output #0, h264, to 'test.h264':
  Stream #0:0: Video: h264, yuv420p(tv, bt709), 1280x720, q=2-31, 6000 kb/s, 25 fps, 25 tbr, 1000k tbn
terminate called after throwing an instance of 'std::runtime_error'
  what():  libav: unable to open output mux for test.h264: Permission denied
Aborted
```
But with the root permission, the live stream H264 works : 
```
$ sudo libcamera-vid --camera 0 --hdr --exposure sport --lens-position 10.2 --nopreview --width 1280 --height 720 --framerate 25 --bitrate 6000000 --codec H264 --timeout 0 --inline --output test.h264
[4:07:32.035802002] [4734]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+118-563cd78e
[4:07:32.050031989] [4737]  INFO RPI pisp.cpp:653 libpisp version v1.0.2 fa44a258644a 22-11-2023 (21:59:22)
[4:07:32.068084968] [4737]  INFO RPI pisp.cpp:1112 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a to CFE device /dev/media0 and ISP device /dev/media2 using PiSP variant BCM2712_C0
[4:07:32.069208305] [4734]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+118-563cd78e
[4:07:32.083162550] [4740]  INFO RPI pisp.cpp:653 libpisp version v1.0.2 fa44a258644a 22-11-2023 (21:59:22)
[4:07:32.101230808] [4740]  INFO RPI pisp.cpp:1112 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a to CFE device /dev/media0 and ISP device /dev/media2 using PiSP variant BCM2712_C0
[4:07:32.102790738] [4734]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format Y16
[... cut ...]
    SRGGB10_CSI2P,1536x864/120.135 - Score: 1100
    SRGGB10_CSI2P,2304x1296/56.0255 - Score: 1400
    SRGGB10_CSI2P,4608x2592/14.3536 - Score: 23592.8
Stream configuration adjusted
[4:19:06.647259934] [5005]  INFO Camera camera.cpp:1183 configuring streams: (0) 1280x720-YUV420 (1) 1536x864-RGGB16_PISP_COMP1
[4:19:06.647330156] [5013]  INFO RPI pisp.cpp:1396 Sensor: /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a - Selected sensor format: 1536x864-SRGGB10_1X10 - Selected CFE format: 1536x864-PC1R

```

Lillifee, 2024-01-01:
Thank you for testing! I'll investigate why this profile setting is no longer supported on the Pi 5. Additionally, I try to remove the profile argument and add it as an option in the web interface. I can't remember the specific reason for adding it, so I'll retest it with the H.264 players.

I'll let you know when i have the new version. Would be great if you could test it once again with the pi5!

Lillifee, 2024-01-02:
Hey @hb9thf 

i removed the --profile baseline from the arguments and added it as a separate setting.
Maybe you could give it a try with the PI 5 and let me know if it works?

Simply install the most recent release.
https://github.com/Lillifee/raspiCam/releases/tag/v2.0.11 

Thanks a lot

hb9thf, 2024-01-02:
Hi @Lillifee 

I did the test, but the libav output context/format seem to be incorrect  : 
```
[server] version 2.0.11
[server] starting services...
[control] starting Stream ...
[process] libcamera-vid --nopreview --width 1280 --height 720 --framerate 25 --bitrate 6000000 --codec H264 --timeout 0 --inline --output -
[button] onoff library not found...
To enable GPIO support please install the onoff library and restart raspiCam. 'npm install onoff'
 Error [ERR_MODULE_NOT_FOUND]: Cannot find package 'onoff' imported from /home/pi/raspiCam/server.js
[server] server listening on 8000
[0:06:16.583445990] [1264]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+118-563cd78e
[0:06:16.597824541] [1271]  INFO RPI pisp.cpp:653 libpisp version v1.0.2 fa44a258644a 22-11-2023 (21:59:22)
[0:06:16.615570848] [1271]  INFO RPI pisp.cpp:1112 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a to CFE device /dev/media3 and ISP device /dev/media0 using PiSP variant BCM2712_C0
[0:06:16.616787786] [1264]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format Y16
[0:06:16.616813361] [1264]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format RGB6
[0:06:16.616820879] [1264]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format BGR6
[0:06:16.616829213] [1264]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format PC1M
Mode selection for 1280:720:12:P(25)
    SRGGB10_CSI2P,1536x864/120.135 - Score: 1100
    SRGGB10_CSI2P,2304x1296/56.0255 - Score: 1400
    SRGGB10_CSI2P,4608x2592/14.3536 - Score: 23592.8
Stream configuration adjusted
[0:06:16.617063585] [1264]  INFO Camera camera.cpp:1183 configuring streams: (0) 1280x720-YUV420 (1) 1536x864-BGGR16_PISP_COMP1
[0:06:16.617130975] [1271]  INFO RPI pisp.cpp:1396 Sensor: /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a - Selected sensor format: 1536x864-SBGGR10_1X10 - Selected CFE format: 1536x864-PC1B
[NULL @ 0x555675704d20] Unable to find a suitable output format for '-'
ERROR: *** libav: cannot allocate output context ***
```

And same result if I run it as root or user...
KR

Lillifee, 2024-01-02:
Hey @hb9thf 

oh noo... I thought the problem is 'only' the profile argument. If you run the same command with an output file it works, right?

I found another forum post regarding the pi5 with camera module 3
https://forums.raspberrypi.com/viewtopic.php?t=360020
https://forums.raspberrypi.com/viewtopic.php?t=361176

I'm really sorry but i'm limited to the libcamera-vid. I hope they can fix the bug with pi5/cam3 soon... 😔

Maybe it's just another missing argument like the libav-format but i can not test it with my setup. If you run the command with --output -, do you get the same error? 
If we can reproduce the error, we could try to play around with the arguments...
--codec libav --libav-format mpegts --libav-audio

hb9thf, 2024-01-02:
Hey @Lillifee 

First of all, I'd like to thank you for your quick replies and for keeping your development up to date.

Yes same error with `--output -` I've the same error : 
```
[NULL @ 0x5556328be080] Unable to find a suitable output format for '-'
ERROR: *** libav: cannot allocate output context ***
```
But I didn't undestand where the output is sent with `--output -` ? Because the same command without `--output` is working well, like this : 
```
$ libcamera-vid --camera 0 --hdr --exposure sport --lens-position 10.2 --nopreview --width 1280 --height 720 --framerate 25 --bitrate 6000000 --codec H264 --timeout 0 --inline
[1:54:10.428087617] [1807]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+118-563cd78e
[1:54:10.442263983] [1810]  INFO RPI pisp.cpp:653 libpisp version v1.0.2 fa44a258644a 22-11-2023 (21:59:22)
[1:54:10.459788722] [1810]  INFO RPI pisp.cpp:1112 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a to CFE device /dev/media3 and ISP device /dev/media0 using PiSP variant BCM2712_C0
[1:54:10.461007927] [1807]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+118-563cd78e
[1:54:10.474931441] [1813]  INFO RPI pisp.cpp:653 libpisp version v1.0.2 fa44a258644a 22-11-2023 (21:59:22)
[1:54:10.493049755] [1813]  INFO RPI pisp.cpp:1112 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a to CFE device /dev/media3 and ISP device /dev/media0 using PiSP variant BCM2712_C0
[1:54:10.493790960] [1807]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format Y16
[1:54:10.493819163] [1807]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format RGB6
[1:54:10.493826645] [1807]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format BGR6
[1:54:10.493835219] [1807]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format PC1M
Mode selection for 1280:720:12:P(25)
    SRGGB10_CSI2P,2304x1296/30.0021 - Score: 1400
Stream configuration adjusted
[1:54:10.494024423] [1807]  INFO Camera camera.cpp:1183 configuring streams: (0) 1280x720-YUV420 (1) 2304x1296-BGGR16_PISP_COMP1
[1:54:10.494127719] [1813]  INFO RPI pisp.cpp:1396 Sensor: /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a - Selected sensor format: 2304x1296-SBGGR10_1X10 - Selected CFE format: 2304x1296-PC1B
[libx264 @ 0x5555696cdb40] using cpu capabilities: ARMv8 NEON
[libx264 @ 0x5555696cdb40] profile High, level 3.1, 4:2:0, 8-bit
Output #0, h264, to '':
  Stream #0:0: Video: h264, yuv420p(tv, bt709), 1280x720, q=2-31, 6000 kb/s, 25 fps, 25 tbr, 1000k tbn
```

Lillifee, 2024-01-02:
Hey @hb9thf 

--output - is sending the stream to the stdout (the application), and from there i stream it to the connected clients on the website. 

You can find it on the documentation page https://www.raspberrypi.com/documentation/computers/camera_software.html

- `-` write to stdout
- udp:// - a string starting with this is taken as a network address for streaming
- tcp:// - a string starting with this is taken as a network address for streaming
- a string containing a %d directive is taken as a file name where the format directive is replaced with a count that increments for each file that is opened. Standard C format directive modifiers are

This works well with all other Pi's until now... 

As far as i understood from the forum entries, it also dont work using an tcp stream as output with pi5/mod3

```libcamera-vid -t 0 --inline --listen -o tcp://0.0.0.0:8554```

Lillifee, 2024-01-03:
Hey @hb9thf 

I received a response from rpicam-apps, and I included the --libav-format h264 as a command line argument. 
Could you please give the new release another attempt?

https://github.com/Lillifee/raspiCam/releases/tag/v2.0.13

hb9thf, 2024-01-03:
Hey @Lillifee 

Unfortunately, this time LIBAV wants a profile baseline....
 ```
$ node server.js
[server] version 2.0.14
[server] starting services...
[control] starting Stream ...
[process] libcamera-vid --nopreview --width 1280 --height 720 --framerate 25 --bitrate 6000000 --profile baseline --codec H264 --libav-format h264 --timeout 0 --inline --output -
[button] onoff library not found...
To enable GPIO support please install the onoff library and restart raspiCam. 'npm install onoff'
 Error [ERR_MODULE_NOT_FOUND]: Cannot find package 'onoff' imported from /home/pi/raspiCam/server.js
[server] server listening on 8000
[0:06:02.631446546] [1220]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+118-563cd78e
[0:06:02.645511774] [1227]  INFO RPI pisp.cpp:653 libpisp version v1.0.2 fa44a258644a 22-11-2023 (21:59:22)
[0:06:02.662890119] [1227]  INFO RPI pisp.cpp:1112 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a to CFE device /dev/media0 and ISP device /dev/media1 using PiSP variant BCM2712_C0
[0:06:02.664143658] [1220]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format Y16
[0:06:02.664181103] [1220]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format RGB6
[0:06:02.664194695] [1220]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format BGR6
[0:06:02.664208992] [1220]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format PC1M
Mode selection for 1280:720:12:P(25)
    SRGGB10_CSI2P,1536x864/120.135 - Score: 1100
    SRGGB10_CSI2P,2304x1296/56.0255 - Score: 1400
    SRGGB10_CSI2P,4608x2592/14.3536 - Score: 23592.8
Stream configuration adjusted
[0:06:02.664510529] [1220]  INFO Camera camera.cpp:1183 configuring streams: (0) 1280x720-YUV420 (1) 1536x864-BGGR16_PISP_COMP1
[0:06:02.664583696] [1227]  INFO RPI pisp.cpp:1396 Sensor: /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a - Selected sensor format: 1536x864-SBGGR10_1X10 - Selected CFE format: 1536x864-PC1B
ERROR: *** libav: no such profile baseline ***

```

Lillifee, 2024-01-03:
Hey @hb9thf 

I apologize for the inconvenience. I attempted using libav on my Pi 4, but unfortunately, the inconsistency persists within libav between Pi 4 and Pi 5. The issue lies in the fact that Pi 5 with libav doesn't support the profile setting.

It's a bit chaotic, in my opinion. I believe the libcamera/rpicam API and results should maintain consistency across different Pi versions. Nevertheless, I proceeded with another release. Could you please give it another try?
 
 https://github.com/Lillifee/raspiCam/releases/tag/v2.0.15

hb9thf, 2024-01-03:
Hey @Lillifee 

No problem, I appreciate your quick support and it only took me a minute or two to test it....
But it's not the main subject, the answer is YES !!! That's work fine : 
```
$ node server.js
[server] version 2.0.15
[server] starting services...
[control] starting Stream ...
[process] libcamera-vid --rotation 180 --nopreview --width 1280 --height 720 --framerate 25 --bitrate 6000000 --codec H264 --libav-format h264 --timeout 0 --inline --output -
[button] onoff library not found...
To enable GPIO support please install the onoff library and restart raspiCam. 'npm install onoff'
 Error [ERR_MODULE_NOT_FOUND]: Cannot find package 'onoff' imported from /home/pi/raspiCam/server.js
[server] server listening on 8000
[0:05:40.652487538] [1263]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+118-563cd78e
[0:05:40.671721174] [1270]  INFO RPI pisp.cpp:653 libpisp version v1.0.2 fa44a258644a 22-11-2023 (21:59:22)
[0:05:40.689826206] [1270]  INFO RPI pisp.cpp:1112 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a to CFE device /dev/media0 and ISP device /dev/media1 using PiSP variant BCM2712_C0
[0:05:40.691955697] [1263]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format Y16
[0:05:40.691984622] [1263]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format RGB6
[0:05:40.691992233] [1263]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format BGR6
[0:05:40.692010344] [1263]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format PC1M
Mode selection for 1280:720:12:P(25)
    SRGGB10_CSI2P,1536x864/120.135 - Score: 1100
    SRGGB10_CSI2P,2304x1296/56.0255 - Score: 1400
    SRGGB10_CSI2P,4608x2592/14.3536 - Score: 23592.8
Stream configuration adjusted
[0:05:40.692281893] [1263]  INFO Camera camera.cpp:1183 configuring streams: (0) 1280x720-YUV420 (1) 1536x864-RGGB16_PISP_COMP1
[0:05:40.692360373] [1270]  INFO RPI pisp.cpp:1396 Sensor: /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a - Selected sensor format: 1536x864-SRGGB10_1X10 - Selected CFE format: 1536x864-PC1R
[libx264 @ 0x5555b51aa530] using cpu capabilities: ARMv8 NEON
[libx264 @ 0x5555b51aa530] profile High, level 3.1, 4:2:0, 8-bit
Output #0, h264, to '-':
  Stream #0:0: Video: h264, yuv420p(tv, bt709), 1280x720, q=2-31, 6000 kb/s, 25 fps, 25 tbr, 1000k tbn
```

[image]

Many Thanks !

Lillifee, 2024-01-03:
Puh perfect, thanks a lot for testing!! 😪 
That's great! I'll close the issue!

closed 2024-01-03 by Lillifee