Lillifee/raspiCam #70: Enhancement request: ROI / "zoom mode"

jil24, 2024-01-20:
Hi! this software is excellent. One enhancement I could use would be to implement something similar the `roi` function from the libcamera demo apps, AKA "digital zoom" (yuck)

from libcamera-still:
`--roi x, y, w, h`

Typically I'd use "10x" specified as `--roi 0.45,0.45,0.1,0.1` to blow up the central region of the image while previewing to adjust fine focus manually (my camera board is hooked up to a microscope). Then I'd take an unzoomed image.

Lillifee, 2024-01-20:
Hey @jil24,

Thanks for your feedback! I've introduced a new setting, "Digital zoom" in the latest release. 
https://github.com/Lillifee/raspiCam/releases/tag/v2.1.1

I called it digital zoom, since it's use only a bare minimum of the ROI function with those presets: 
- 1x - no zoom (0,0,1,1)
- 2x - 0.25,0.25,0.5,0.5
- 5x - 0.4,0.4,0.2,0.2
- 10x - 0.45,0.45,0.1,0.1

You can find the new setting in the camera options as a slider:
[image]

Let me know if it works for you.

jil24, 2024-01-21:
It seems to work quite well!

Lillifee, 2024-01-22:
aperfect, happy to hear that 😉👍

closed 2024-01-22 by Lillifee