Lillifee/raspiCam #71: ERROR: *** unrecognised option '--libav-format' ***

tourbino, 2024-03-03:
I'm running raspiCam on a rpi 4b with HQ cam. When starting node.js I get the following:

```
~/raspiCam`` $ sudo node server.js
[server] version 2.1.0
[server] starting services...
[control] starting Stream ...
[process] libcamera-vid --nopreview --width 1280 --height 720 --framerate 25 --bitrate 6000000 --codec H264 --libav-format h264 --timeout 0 --inline --output -
[button] onoff library not found...
To enable GPIO support please install the onoff library and restart raspiCam. 'npm install onoff'
 Error [ERR_MODULE_NOT_FOUND]: Cannot find package 'onoff' imported from /home/pi/raspiCam/server.js
ERROR: *** unrecognised option '--libav-format' ***
[server] server listening on 8000 
```

I've read that the --libav format is from the full/desktop version of Raspberry Pi OS. What can be done about this?

Lillifee, 2024-03-03:
Hey @tourbino 

strange, i have also a pi4 with hq and lite os. did you update your raspberry pi to the latest version?

sudo apt update
sudo apt upgrade

I added some libav settings recently to support the pi5.

tourbino, 2024-03-03:
Strange. Yes, I started a fresh 64bit fresh lite install and updated and upgraded everything.

tourbino, 2024-03-03:
Btw - when I use libcamera-still I get a proper image.

Lillifee, 2024-03-04:
Hey @tourbino,

I'm going to try installing a fresh 64-bit version the next days and give it a go. 
Also, I've made a new release where you can now set the libav format to none. Since this is necessary for the PI5, I'm keeping the default h264 for now. Hopefully, they'll standardize the arguments across libcamera for pi4, pi5, lite, etc. If not, I'll have to add presets for different platforms and hardware :( 

[image]

Could you please test if the new release, with the libav format set to none, works for you?
https://github.com/Lillifee/raspiCam/releases/tag/v2.1.2

tourbino, 2024-03-04:
Thank you very much! I tried the new version, and now it works when I select the "none" format. Still the same error message in console though. You have made a great tool, thank you again.

closed 2024-03-04 by tourbino