Lillifee/raspiCam #77: libcamera-vid *** unrecognised option '--libav-format' ***

jsmr1, 2024-09-05:
Hi, I try to install the package with a fresh and up to date raspi-os-bookworm-arm64 with the 
tar ball and the clone method but I get always the error

```
root@raspberrypi:/usr/local/src2/raspiCam/build>node server.js
[server] version 2.1.2
[server] starting services...
[control] starting Stream ...
[process] libcamera-vid --nopreview --width 1280 --height 720 --framerate 25 --bitrate 6000000 --codec H264 --libav-format h264 --timeout 0 --inline --output -
ERROR: *** unrecognised option '--libav-format' ***
[server] server listening on 8000
[control] starting Stream ...
[process] libcamera-vid --width 1280 --height 720 --framerate 25 --bitrate 6000000 --codec H264 --libav-format h264 --timeout 0 --inline --output -
ERROR: *** unrecognised option '--libav-format' ***

```

libcamera-vid don't know about the option "--libav-format". 

`libcamera-vid --width 1280 --height 720 --framerate 25 --bitrate 6000000 --codec H264  --timeout 0 --inline --output -`

runs fine.

I have seen this has addressed in the recent past. But I don't know to suppress this option. 

Juergen

jsmr1, 2024-09-05:
Ok, get the patch. Its 10 years ago I used the last time a diff for that. Here is the patch:

in **shared/settings/defaultSettings.ts**:

```
  stream: {
    width: streamSettingDesc.width.defaultValue,
    height: streamSettingDesc.height.defaultValue,
    framerate: streamSettingDesc.framerate.defaultValue,
    codec: streamSettingDesc.codec.defaultValue,
    bitrate: streamSettingDesc.bitrate.defaultValue,
//    'libav-format': streamSettingDesc['libav-format'].defaultValue,
```

in **shared/settings/stream.ts**
 
```
  /**
   * Set the libav output format to use. These output formats can be specified as containers (e.g. mkv, mp4, avi)
   * or stream output (e.g. h264 or mpegts). If this option is not provided, libav tries to deduce the output format from the filename specified by the -o command line argument.
   */
//  'libav-format': enumSetting('Libav format', ['none', 'h264'], 'h264'), 

[..]

  /** if (passThroughSettings['libav-format'] === 'none') {
      passThroughSettings['libav-format'] = undefined;  */
  } 

```

Lillifee, 2024-09-06:
Hey @jsmr1 

It's unfortunate that not all arguments work on the different Pi devices. That's why I provided the option to disable the libav format. The error only affects the stream, but the website functions correctly, so no code adjustments are necessary. You can simply navigate to the website and set the libav format to 'none.'

However, I will keep the bug open and look into improving the initial startup, or possibly create a setup wizard to configure the settings on the first run.

Thank you for your response, and I apologize for any inconvenience

[image]

jsmr1, 2024-09-06:
Yes, you are right, I got it. Next problem I had had was to set the output to mjpeg on the website to get an output on http://<IP>:8000/api/stream/mjpeg. (Please  show the urls on the website, maybe make a button to switch directly for navigation, this would make the search in the documentation unnecessary.)

Your free software works very fine, thank you for your work.

Lillifee, 2024-09-06:
Initially, the MJPEG stream wasn't intended for use outside the app, but that's a great idea. Thanks a lot!

mbarela82, 2025-05-22:
Hello, I just installed the raspiCam and I'm getting the same error.