Lillifee/raspiCam #79: errors in stream

crepp1967, 2024-11-08:
I probably have a configuration problem, as soon as I start the node.js server I get error messages and the image doesn't look so great either
[image]

Here is a look into the logfile
`[server] server listening on 8000
[117:56:28.574007020] [16910]  INFO Camera camera_manager.cpp:297 libcamera v0.0.5+83-bde9b04f
[117:56:28.709279283] [16916]  INFO RPI vc4.cpp:437 Registered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device /
[117:56:28.709372152] [16916]  INFO RPI pipeline_base.cpp:1101 Using configuration file '/usr/share/libcamera/pipeline/rpi
Mode selection:
    SRGGB10_CSI2P 1536x864 - Score: 3100
    SRGGB10_CSI2P 2304x1296 - Score: 3400
    SRGGB10_CSI2P 4608x2592 - Score: 25592.8
[117:56:28.719099001] [16910]  INFO Camera camera.cpp:1033 configuring streams: (0) 1280x720-YUV420 (1) 1536x864-SBGGR10_C
[117:56:28.719869233] [16916]  INFO RPI vc4.cpp:565 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 15
[117:56:31.654354225] [16922] ERROR IPARPI cam_helper_imx708.cpp:273 PDAF data in unsupported format
[117:56:31.654718146] [16922] ERROR IPARPI cam_helper_imx708.cpp:341 Lin/Log histogram mismatch
[117:56:31.854314558] [16922] ERROR IPARPI cam_helper_imx708.cpp:273 PDAF data in unsupported format
[117:56:36.932848717] [16922] ERROR IPARPI cam_helper_imx708.cpp:273 PDAF data in unsupported format
[117:56:39.771758303] [16922] ERROR IPARPI cam_helper_imx708.cpp:273 PDAF data in unsupported format
[117:56:40.211632132] [16922] ERROR IPARPI cam_helper_imx708.cpp:273 PDAF data in unsupported format
[117:56:40.771437453] [16922] ERROR IPARPI cam_helper_imx708.cpp:273 PDAF data in unsupported format`

What con i do to fix it?

Lillifee, 2024-11-08:
Hey @crepp1967,

Unfortunately, those errors are coming from `libcamera-vid`. Did you adjust any settings, or has this been happening from the beginning? In your log, the specific `libcamera-vid` command and its arguments are missing, which might help us diagnose the issue.

Could you try running `libcamera-vid` without any custom settings, then add each argument one at a time to pinpoint which setting might be causing the error? 
```
libcamera-vid -t 10000 -o test.h264
```
Also, I came across this link—could you check the cable as well?
- https://forums.raspberrypi.com/viewtopic.php?t=350277
- https://github.com/raspberrypi/libcamera/issues/61

crepp1967, 2024-11-10:
i will check the cabel next week an will reporting here

crepp1967, 2024-11-12:
Now i have checked the cabel and move the pi to an other place and all is fine 👍 
thanks for helping tipps

closed 2024-11-12 by Lillifee