Lillifee/raspiCam #83: Camera maximum resolution modification

eva-00100, 2025-05-21:
Hello, I used a higher resolution camera to take photos and videos, but the highest resolution in the settings is fixed. I would like to ask how I can modify the code.

eva-00100, 2025-05-21:
My sensor is imx283.

Lillifee, 2025-05-21:
Hey @eva-00100,

I've increased the maximum photo resolution to 20.2 MP (5496 × 3672).
I haven't had a chance to test it yet... Could you give it a try and let me know if everything works as expected?
If all looks good, I'll go ahead and create a release.

[raspiCam.zip](https://github.com/user-attachments/files/20371194/raspiCam.zip)
[raspiCam.tar.gz](https://github.com/user-attachments/files/20371193/raspiCam.tar.gz)

[image]

Lillifee, 2025-05-21:
Really nice!! Thanks a lot!
Please keep me updated on your project and the 3D model.

eva-00100, 2025-05-22:
First, it is recommended to adjust the shutter speed to be more refined, and second, whether it can add the function of adjusting the ISO❤️❤️❤️

Lillifee, 2025-05-22:
I started this project a few years ago using the old raspistill and raspivid camera apps. Back then, ISO could be set directly. Since then, Raspberry Pi switched to using the libcamera apps, which only expose an "exposure" property. I haven’t had much time to work on it recently, and now it looks like they’re switching again—this time to the rpicam apps, which are also based on libcamera.

I think I’ve covered most of the available settings, but if you notice any missing flags, feel free to let me know.
https://www.raspberrypi.com/documentation/computers/camera_software.html

Aside from that, there’s currently no localization support. It would need to be implemented, but at the moment, I don’t have enough time to start a completely new feature for raspicam.

If the resolution works well, I'll go ahead and release it and close this issue. If you have a new topic, please create a separate issue for it. I’ll create a new one for the localization.

eva-00100, 2025-05-22:
Okay, thank you. The problem has been resolved.

closed 2025-05-22 by eva-00100