Lillifee/raspiCam #85: button to trigger the problem

eva-00100, 2025-05-23:
I found that I was prompted to install "onoff" when I was using it, but I already installed it 

[image]

eva-00100, 2025-05-23:
[image]

eva-00100, 2025-05-23:
The board I'm using is "pi5"
The image below is what I got after typing the command

[image]
I tried modifying the button.json

[image]
rerun

[image]
This time there was no error, but I still can't use the button control

eva-00100, 2025-05-23:
I used a python script to verify the integrity of my button installation

[image]

Lillifee, 2025-05-23:
Hmm...
Is there an info log statement like setup gpio pin 588 during the startup of raspicam?
It’s possible that the pin isn't even being registered—this range might not have been supported previously. :(

eva-00100, 2025-05-23:
Indeed, I did not see any relevant log statements.❤️😂😂[image]

Lillifee, 2025-05-24:
Hey @eva-00100 

I’ve released a new version with updated GPIO addresses.
All packages have been upgraded to the latest versions, and I'm now using rpicam-vid and rpicam-still.

https://github.com/Lillifee/raspiCam/releases/tag/v2.2.0

[image]

I wasn't able to test it with the Raspberry Pi hardware, as I don’t currently have it set up and wired. 
However, everything looks good so far.

eva-00100, 2025-05-24:
Sorry, I replied late, I tested it and I got the following error
[image]

Lillifee, 2025-05-24:
onoff is using the sysfs interface for GPIO access. I'm not sure if this interface is still supported on the Raspberry Pi 5. Unfortunately, I don’t have a Pi 5 to test it myself.

Could you check if the /sys/class/gpio directory exists on your setup?

The only relevant info I found was this forum thread:
https://forums.raspberrypi.com/viewtopic.php?t=359302

hmm...

Lillifee, 2025-05-24:
I'm really sorry...

It looks like I’ll need to switch to a different approach than sysfs with the onoff library. I’ll have to explore which libraries are available, as the entire sysfs is going to be deprecated. 😔

eva-00100, 2025-05-24:
Thank you very much for helping me fix these features, and now it's also very good and I'm sure you'll be improving them and making it even better in the near future 😍💕💕😁👍👍👍👍👍👍

eva-00100, 2025-05-24:
Thank you very much for helping me fix these features, and now it's also very good and I'm sure you'll be improving them and making it even better in the near future
👍👍👍👍👍👍😁💕💕

Lillifee, 2025-05-25:
Hey @eva-00100,
I just noticed that the address you’re using is different from the one I copied. 
Could you try setting it to 588 in the settings file? It should be accepted since the latest release. 
For me it was 529 - GPIO17... for yours its 588 - GPIO17
Looks like the mapping is different on the Pi 5.

Lillifee, 2025-05-25:
I’ve created another release that includes a number input:
[image]

This should be the easiest version to use for now, at least until I have some time to improve it further.

https://github.com/Lillifee/raspiCam/releases/tag/v2.2.1

Thanks a lot for testing!!

eva-00100, 2025-05-25:
Thank you for the improvements you have made to this project 😊， In the future, I will inform you of the progress of my camera as soon as possible 📷